wpietri/sucks #1: Getting recipient-unavailable reply from Ecovacs API

Zenconomy, 2017-12-01:
Hey! Thanks for taking your time to put this project together.

I just downloaded and tried it out. I'm in Canada so had to change country to ca, then I was able to login. Once logged in, if I try to run any command, it stops with the response below.

I have the Deebot M81 pro which I suppose could be the cause.

DEBUG    SEND: <iq id="6a6842cd-e361-4f9c-8716-7331c3258003-3" to="E0000700917605281326@126.ecorobot.net/atom" from="201712015a219468a2480@ecouser.net/49116f9d9e23a3f1618d51bdd194fed8" type="set"><query xmlns="com:ctl"><ctl td="Charge"><charge type="go" /></ctl></query></iq>

DEBUG    RECV: <iq type="error" to="201712015a219468a2480@ecouser.net/49116f9d9e23a3f1618d51bdd194fed8" from="E0000700917605281326@126.ecorobot.net/atom" id="6a6842cd-e361-4f9c-8716-7331c3258003-3"><error type="wait" code="404"><recipient-unavailable xmlns="urn:ietf:params:xml:ns:xmpp-stanzas" /></error></iq>

I'm attaching the complete debug output as well. Any ideas? Thanks :)

[sucks-debug.txt](https://github.com/wpietri/sucks/files/1522834/sucks-debug.txt)

wpietri, 2017-12-01:
Hello! I'm excited that you tried it. I think the protocols for the robots must be different. That charge command works for my robot.

If you're feeling bold, I'd suggest you use [xmppproxy](https://www.beneaththewaves.net/Software/XMPPPeek.html) to snoop what's going on between your app and the robot. If you send me a dump, it shouldn't be too hard for me to make it work for both.

Zenconomy, 2017-12-02:
I used mitmproxy to listen to the IOS app. I only see one request to the server to load the list of available robots, POST to https://47.88.66.164:8000/user.do

If I turn mitmproxy off, the app can't find the robot.

However, once I select the robot in the app, commands between the app and the robot aren't being picked up by mitmproxy at all. In fact, I can turn mitmproxy off and the app will still control the robot... even though the phone's browser can't connect to the internet because the proxy is down.

Ideas?

wpietri, 2017-12-02:
Yes. Alas, you'll also have to set up [xmppproxy](https://www.beneaththewaves.net/Software/XMPPPeek.html), which is a different tool entirely. As described in [my protocol docs](https://github.com/wpietri/sucks/blob/master/protocol.md), the vaccum and the app communicate via XMPP. (Maybe mitmproxy also does XMPP; I didn't try it.)

Zenconomy, 2017-12-02:
Thanks, sorry I should have looked at the docs first.

How did you configure xmppproxy to relay traffic from the app? Custom routes on your router?

wpietri, 2017-12-03:
I just told my router to say that the IP for msg-na.ecouser.net was for my laptop. Then I ran the proxy like this:

```python ./xmpppeek.py  47.88.66.164 cert.pem key.pem```

That IP address being the the one for msg-na.ecouser.net, of course.

AbsoluteCatalyst, 2017-12-03:
Hello! Thank you for putting this all together. I own a Deebot M80 Pro I purchased on sale and was trying to get this code to work for me as well, and I've figured it out! 

The JSON string returned via __call_user_api actually contains the model of your deebot as the variable "class"! 
For example: 
126 is the Deebot N79 
141 is the Deebot M81 Pro
125 is the Deebot M80 Pro
If we feed this model number back into the iq_query's URL (under wrap_command) Eg:

> VacuumID@**125**.ecorobot.net/atom (for my M80 Pro)
> VacuumID@**141**.ecorobot.net/atom (for **your** M81 Pro)

then it works and correctly controls your specific Deebot! 
I'm pretty sure 141 is the class number for the M81 Pro as someone else has intercepted it's incoming traffic as well. See: https://robert.penz.name/1462/a-security-minded-guy-forced-to-buy-a-wifi-enabled-cleaning-robot/
Difference being he owns an M81 Pro, and looking at his traffic it's almost the same except for that one small difference in the sub-domain. 

The change to the python to make this work for all bots is around 2-3 lines. Just store the value returned by "class", and append it to the URL string. 

This is exciting! I've loved this cute vacuum ever since I've gotten it.

wpietri, 2017-12-03:
Excellent detective work, @AbsoluteCatalyst! Let me rejig things to support models you describe here, plus different countries and regions. Then folks won't have to edit the python just to get it work.

wpietri, 2017-12-03:
I'd possibly added support for different models, as suggested by @AbsoluteCatalyst. If folks could pull this code and try it out, I'd appreciate it.

Also, if it's handy, I'd love to know what domain names people are seeing the existing app use in various places so I can make that properly configurable. Are you seeing things like eco-{country}-api.ecovacs.com and users-{continent}.ecouser.net? Are there any other location-specific changes that you're making?

r3dsub, 2017-12-03:
Class 107 for Model M88. Could not test it yet but sounds promising.

Zenconomy, 2017-12-03:
I pulled the lastest repo, changed meta.country to 'ca' and logged in. Now works perfectly with my M81 pro. Awesome work @wpietri!

May I suggest putting the country variable in the config file and prompt the user to choose country when they run the login command?

Thanks for the detective work @AbsoluteCatalyst, spared me having to fiddle with xmpppeek.py :)

wpietri, 2017-12-04:
Exactly my thinking too. Login now prompt for both country code and continent code, and tries to make intelligent guesses for each.

If non-US folks could try this out and let me know how it works for them, I'd appreciate it. Just back up your config, do "sucks login", and follow the prompts. Hopefully this will work for at least the people who've described their setups so far.

Also, since it appears that people are using this, I have created a mailing list:

https://groups.google.com/forum/#!forum/sucks-users

People should feel free to use it as they see fit, but I'll at least be announcing new versions and asking occasional questions. Please do sign up so that we can keep the conversation once I close this issue!

Zenconomy, 2017-12-04:
This is what I get when running the login command in the latest version:

Ecovacs app email: torbjornaxelson@gmail.com
Ecovacs app password: 
your two-letter country code [ca]: ca
your two-letter continent code [na]: na
Traceback (most recent call last):
  File "sucks.py", line 435, in <module>
    cli()
  File "/Users/T/.local/share/virtualenvs/sucks-nR-cWLjn/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/Users/T/.local/share/virtualenvs/sucks-nR-cWLjn/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/Users/T/.local/share/virtualenvs/sucks-nR-cWLjn/lib/python3.6/site-packages/click/core.py", line 1092, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/Users/T/.local/share/virtualenvs/sucks-nR-cWLjn/lib/python3.6/site-packages/click/core.py", line 895, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/Users/T/.local/share/virtualenvs/sucks-nR-cWLjn/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "sucks.py", line 367, in login
    EcoVacsAPI(device_id, email, password_hash)
TypeError: __init__() missing 2 required positional arguments: 'country' and 'continent'

wpietri, 2017-12-04:
Ah, my mistake. Sorry! Clearly I need some automated acceptance tests for things like login. Please give it a try again.

Zenconomy, 2017-12-04:
Works with your latest commit! I would call this issue resolved, at least for my case (Canada, M81 pro)

wpietri, 2017-12-04:
Thanks! Glad to hear it. I'll leave this open for a couple of days to see how it works for the other folks, but I think it should be fine for them.

r3dsub, 2017-12-04:
great work! didn't change anything now (just path to the config file because running this on windows currently). and it just works now. I'm from germany running a Deebot M88.
next step for me is to integrate this in my homematic.

wpietri, 2017-12-04:
Great to hear. @r3dsub, where is the right place on Windows to put a config file? I'm glad add support for that.

wpietri, 2017-12-04:
It looks like we have basic support for more robots and locations, so I'm closing this. If interested parties could join the mailing list to carry on the conversation, I'd appreciate it:

https://groups.google.com/forum/#!forum/sucks-users

closed 2017-12-04 by wpietri