wpietri/sucks #2: RESTful API

Zenconomy, 2017-12-04:
It would be really neat to have a simple REST server on top of sucks to simplify integration with other apps and services.

I'd be happy to help with implementation but I'm more familiar with node than python so probably better if I stick to testing for now :)

wpietri, 2017-12-04:
This is a fine idea. I don't personally have a need for it, but if somebody gets excited I don't think it would be too hard to do. I could pull the existing core objects apart from the command-line script. Then someone could use something like Flask to make a simple API server that called the VacBot object.

If you (or anybody) want to take a swing at it, let me know.

Out of curiosity, what's your use case for the API? Integrating with something else?

Zenconomy, 2017-12-04:
Use case would be to integrate with other workflows so for example being able to control the robot through a service like IFTTT or integrate with Home Assistant.

If you wanna separate the core functionality and the CLI that would be awesome, then I can implement the REST server.

wpietri, 2017-12-04:
This separation is now complete. The CLI-specific code is now in sucks/cli.py. I'd be open to having something like sucks/rest.py if you wanted to include it in this project, and you could of course just use this as a library and put up your own project. I'd say tinker with it for a while and let me know what you think.

elijahnyp, 2018-07-28:
Thanks for separating the api from the cli.  I’ve had to make a few modifications to get it working with my n79s the way I wanted but it gave me the core capabilities and structure. I went ahead and wrote full server that maintains a permanent connection to the vac for very quick interactions and to provide a webapi for status and commands.  It even sends pushover alerts for errors!  I have it integrated into openHAB via mqtt as well.
Thanks again for all your work!
If somebody finds this and wants the code, just let me know and I’ll move it up my priority list.  It’s not clevaned up yet and has a bunch of hard-coded config I’ll need to extract before I make it public.

arid154, 2018-08-06:
@elijahnyp
i'd love to get hands an your method. i have currently IoBroker running and want to implement my deebot 2 into my graphical interface. any chance you can help me with that?

guillebot, 2018-11-23:
I definitely would like to take a look at your code.

I'm in the same path here, integrating sucks with openhab via mqtt.

> Thanks for separating the api from the cli. I’ve had to make a few modifications to get it working with my n79s the way I wanted but it gave me the core capabilities and structure. I went ahead and [...]