wpietri/sucks #4: Feature idea: Move/Rotate deebot for a certain amount of time.

AbsoluteCatalyst, 2017-12-04:
I'm not sure if this might be out of the scope of this project, but it'd be pretty useful to be able to move/rotate the bot itself for a certain amount of time. 

What's interesting right now is that if you send the command "TurnLeft" or "TurnRight" the bot will just rotate in a complete 360 degree fashion. I'm assuming that the app itself only sends a subsequent stop command right as you let go of the button itself. (I haven't taken the time to intercept the connections from the app myself yet unfortunately.)

While this might not be of much use for this project's original goals, it might be useful as a foundation for other people to work from and create some nifty and elaborate projects. Maybe even some more advanced pathfinding/ai logic when it's commanded to go to home base. 

I can work on some python for this idea throughout the week and post it here when I get a chance.

wpietri, 2017-12-04:
Thanks for mentioning this.

I want to break the code out into two chunks: a library that will let others talk to the robot, and the command-line script suitable for cron-like use. It sounds like for your use case, you're mainly thinking about the former.

I'm fine with the library implementing commands like TurnLeft. As part of #3, I'll be breaking out the library soon, which would make it easier for you to submit a pull request that adds commands that don't make sense in a CLI context.

Then rather than hacking on the existing script, you can just do something like:

```
include sucks

api = EcoVacsAPI(config['device_id'], config['email'], config['password_hash'],
                         config['country'], config['continent'])
my_vac = api.devices()[0]
vacbot = VacBot(api.uid, api.REALM, api.resource, api.user_access_token, my_vac, config['continent'])
vacbot.connect_and_wait_until_ready()

vacbot.run(TurnLeft())
sleep(0.1)
vacbot.run(Stop())
```

Would that be enough to get you started on your experiments?

AbsoluteCatalyst, 2017-12-04:
Yes, an API such as that would be good. Though I'd imagine it'd need a persistent connection to the deebot for it to work fluidly. Someone else mentioned this idea already in another issue post though, so that'd be the first step. 

(Also as a side note for the description of the project, this library also works for the M80 Pro Deebot such as mine.)

wpietri, 2017-12-04:
Ok. I have completed the changes that should allow you to experiment along these lines. Let me know what you discover, and I'm glad to discuss pull requests once you have something you like.

Zenconomy, 2017-12-11:
Pull request #18 contains VacBotCommands that allow for this. To use:

vacbot.run(Move('SpinRight'))
time.sleep(0.2)
vacbot.run(Move('stop'))

Valid move actions (that we know of):  forward, SpinLeft, SpinRight, stop, TurnAround

We could abstract further but I think it might be better to just use the native action values as other robot models may implement more actions.

wpietri, 2017-12-19:
I believe @torbjornaxelsson's latest pull request completes this enhancement.

closed 2017-12-19 by wpietri