wpietri/sucks #5: Add support for Windows

wpietri, 2017-12-04:
The script is [reported to work](https://github.com/wpietri/sucks/issues/1#issuecomment-349008481) on Windows except for the config file location. That should be fixed.

r3dsub, 2017-12-04:
I only hardcoded the Windows Path c:\\users\\... \\sucks.Cfg to the config file with double backslashs to get the config file to be found. I think the tilde in Windows is just a dot. But Im not Sure about that. Also as mentioned the crypto module has to be changed to cryptodome under Windows.

wpietri, 2017-12-04:
Ok. Reading around, it looks like I could drop the config file in the %%APPDATA%% or %%LOCALAPPDATA%% directory. Or I could create a `.config` directory in the user's home dir. Which would you prefer?

r3dsub, 2017-12-04:
I would say %appdata% seems the right place. :)

wpietri, 2017-12-05:
I think I have a fix for the config file path. But I can't reproduce the error with the crypto stuff. E.g., when I run the tests under Windows, they're fine:

https://ci.appveyor.com/project/wpietri/sucks/build/1.0.12

Could you reply with a) the error that happens with unmodified code, and b) your fix? Thanks!

wpietri, 2017-12-11:
Having not heard back on the crypto stuff, I've at least merged the config file changes, which should go out with the next release.

wpietri, 2018-07-17:
Having not heard back, I'm going to assume this works. If not, anyone should feel free to reopen the issue.

closed 2018-07-17 by wpietri