wpietri/sucks #11: Problem with Connection

steveknoefler, 2017-12-08:
Hi,
Thank you so much for your ecovacs script.
Unfortunately I have a problem with the connection setup. can you please take a look at it? 

thank you

DEBUG    Event triggered: ssl_cert
ERROR    Connection error.
ERROR    <TagSet object at 0x7fd848a31d30 tags 0:32:16> not in asn1Spec: <OctetString schema object at 0x7fd8481cc898 encoding iso-8859-1 tagSet <TagSet object at 0x7fd84b9849b0 tags 0:0:4>>
Traceback (most recent call last):
  File "/usr/local/lib/python3.5/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1492, in _process
    if not self.__read_xml():
  File "/usr/local/lib/python3.5/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1564, in __read_xml
    self.__spawn_event(xml)
  File "/usr/local/lib/python3.5/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1632, in __spawn_event
    handler.prerun(stanza_copy)
  File "/usr/local/lib/python3.5/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
    self.run(payload, True)
  File "/usr/local/lib/python3.5/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/local/lib/python3.5/dist-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
    if self.xmpp.start_tls():
  File "/usr/local/lib/python3.5/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 889, in start_tls
    cert.verify(self._expected_server_name, self._der_cert)
  File "/usr/local/lib/python3.5/dist-packages/sleekxmpp/xmlstream/cert.py", line 134, in verify
    cert_names = extract_names(raw_cert)
  File "/usr/local/lib/python3.5/dist-packages/sleekxmpp/xmlstream/cert.py", line 73, in extract_names
    asn1Spec=OctetString())[0]
  File "/usr/local/lib/python3.5/dist-packages/pyasn1/codec/ber/decoder.py", line 1318, in __call__
    '%s not in asn1Spec: %r' % (tagSet, asn1Spec)
pyasn1.error.PyAsn1Error: <TagSet object at 0x7fd848a31d30 tags 0:32:16> not in asn1Spec: <OctetString schema object at 0x7fd8481cc898 encoding iso-8859-1 tagSet <TagSet object at 0x7fd84b9849b0 tags 0:0:4>>
DEBUG    reconnecting...
DEBUG    Event triggered: session_end
DEBUG    SEND (IMMED): </stream:stream>
INFO     Waiting for </stream:stream> from server
DEBUG    Event triggered: disconnected
DEBUG     ==== TRANSITION connected -> disconnected
DEBUG    connecting...
DEBUG    Waiting 2.1977662435578362 seconds before connecting.
DEBUG    No remaining DNS records to try.
DEBUG    Waiting 3.661543940220445 seconds before connecting.

wpietri, 2017-12-08:
Hi, Steve. At first guess, I'd think this a network issue. Does the smartphone app work from the exact same network connection? If so, please attach a full debug log. Thanks!

steveknoefler, 2017-12-08:
yes it works
where can I find the debug file or do you mean python3 sucks.py --debug clean ??

steveknoefler, 2017-12-09:
root@nd-sv-oh:/etc/openhab2/sucks-0.8.1# sudo python3 sucks.py --debug clean 10
DEBUG    will run [<__main__.Clean object at 0x7f7ff8be9a58>, <__main__.Charge object at 0x7f7ff8be9ac8>]
DEBUG    Setting up EcoVacsAPI
DEBUG    calling main api user/login with (('account', XXX'), ('password', XXXX'))
INFO     Starting new HTTPS connection (1): eco-de-api.ecovacs.com
DEBUG    "GET /v1/private/de/en/0c06781391c3fece1d90bb002d541a94/i_eco_e/1.3.5/c_googleplay/1/user/login?account=XXXX%2F5P5%XXXX%2BmSPffV7KXW%2FEV3ykXzlBxAtbzEExJUrW3QHma5ruZpKR9segYGN2C0GKfAYvI%3D&password=XXXX&requestId=4998380765d59483c8795f5231c6d339&authTimespan=1512821558619&authTimeZone=GMT-8&authAppkey=eJUWrzRv34qFSaYk&authSign=XXXX HTTP/1.1" 200 None
DEBUG    got {'code': '0000', 'data': {'username': None, 'email': 'mail@mail.com', 'country': 'de', 'accessToken': 'XXX', 'uid': '201711155a0c810628bdc'}, 'msg': '操作成功', 'time': 1512821558957}
DEBUG    calling main api user/getAuthCode with (('uid', 'XXXX'), ('accessToken', 'XXX'))
INFO     Starting new HTTPS connection (1): eco-de-api.ecovacs.com
DEBUG    "GET /v1/private/de/en/XXX/i_eco_e/1.3.5/c_googleplay/1/user/getAuthCode?uid=XXX&accessToken=XXX&requestId=0a672108cac3a338af7c16034d941688&authTimespan=1512821559050&authTimeZone=GMT-8&authAppkey=eJUWrzRv34qFSaYk&authSign=642f91929f3eb51ed965b560a32abf5f HTTP/1.1" 200 None
DEBUG    got {'code': '0000', 'data': {'authCode': 'XXXX'}, 'msg': '操作成功', 'time': 1512821559089}
DEBUG    calling user api loginByItToken with {'userId': 'XXXX', 'token': 'XXXX', 'country': 'DE', 'resource': '0c067813', 'realm': 'ecouser.net'}
INFO     Starting new HTTPS connection (1): users-eu.ecouser.net
DEBUG    "POST /user.do HTTP/1.1" 200 129
DEBUG    got {'result': 'ok', 'todo': 'result', 'resource': '0c067813', 'userId': 'XXXX', 'token': 'XXXX'}
DEBUG    EcoVacsAPI connection complete
DEBUG    calling user api GetDeviceList with {'userid': 'XXXX', 'auth': {'resource': '0c067813', 'userid': 'XXXX', 'token': 'XXXX', 'realm': 'ecouser.net', 'with': 'users'}}
INFO     Starting new HTTPS connection (1): users-eu.ecouser.net
DEBUG    "POST /user.do HTTP/1.1" 200 164
DEBUG    got {'result': 'ok', 'todo': 'result', 'devices': [{'company': 'eco', 'did': 'XXXX', 'name': 'XXXX', 'resource': 'atom', 'nick': None, 'class': '115'}]}
DEBUG    Loaded Plugin: RFC 6120: Stream Feature: STARTTLS
DEBUG    Loaded Plugin: RFC 6120: Stream Feature: Resource Binding
DEBUG    Loaded Plugin: RFC 3920: Stream Feature: Start Session
DEBUG    Loaded Plugin: RFC 6121: Stream Feature: Roster Versioning
DEBUG    Loaded Plugin: RFC 6121: Stream Feature: Subscription Pre-Approval
DEBUG    Loaded Plugin: RFC 6120: Stream Feature: SASL
WARNING  fulljid property deprecated. Use boundjid.resource
DEBUG    Waiting 2.114340627358512 seconds before connecting.
DEBUG    DNS: Querying msg-eu.ecouser.net for AAAA records.
DEBUG    DNS: Error retreiving AAAA address info for msg-eu.ecouser.net.
DEBUG    DNS: Querying msg-eu.ecouser.net for A records.
DEBUG    Connecting to 47.91.67.63:5223
DEBUG    Event triggered: connected
DEBUG     ==== TRANSITION disconnected -> connected
DEBUG    Starting HANDLER THREAD
DEBUG    Loading event runner
DEBUG    SEND (IMMED): <stream:stream to='ecouser.net' xmlns:stream='http://etherx.jabber.org/streams' xmlns='jabber:client' xml:lang='en' version='1.0'>
DEBUG    RECV: <stream:stream from="ecouser.net" id="XXX" version="1.0">
DEBUG    RECV: <stream:features xmlns="http://etherx.jabber.org/streams"><auth xmlns="http://jabber.org/features/iq-auth" /><starttls xmlns="urn:ietf:params:xml:ns:xmpp-tls"><required /></starttls><mechanisms xmlns="urn:ietf:params:xml:ns:xmpp-sasl"><mechanism>PLAIN</mechanism></mechanisms></stream:features>
DEBUG    SEND (IMMED): <starttls xmlns="urn:ietf:params:xml:ns:xmpp-tls"><required /></starttls>
DEBUG    RECV: <proceed xmlns="urn:ietf:params:xml:ns:xmpp-tls" />
DEBUG    Starting TLS
INFO     Negotiating TLS
INFO     Using SSL version: TLSv1
DEBUG    CERT: -----BEGIN CERTIFICATE-----
MIIFajCCBFKgAwIBAgIQYgA2/ylUOftKo8IRAhktgzANBgkqhkiG9w0BAQsFADBC
MQswCQYDVQQGEwJVUzEWMBQGA1UEChMNR2VvVHJ1c3QgSW5jLjEbMBkGA1UEAxMS
UmFwaWRTU0wgU0hBMjU2IENBMB4XDTE2MTIxNzAwMDAwMFoXDTE3MTIxNzIzNTk1
OVowGDEWMBQGA1UEAwwNKi5lY291c2VyLm5ldDCCASIwDQYJKoZIhvcNAQEBBQAD
ggEPADCCAQoCggEBALy4afVdU9lyFERfL03tK6dqRLBlZKlUEVLDeN07uDnULq3M
YJ/nPsb/ar81gprQTQ56ilV4dDOcaZ+LWfRo4PV4UHLMULJgmk7pxcTEFI4cuoDX
ptOwAEIycc92ikLFjWYY+bN6BwVMD5mSmIR6iVCgda8lc9Sm8ES+nv2wLD5pL3Pw
3TuqcXCT9FTa2PpFYgI1Rw8JNX7wu7MGRaZfFibuWgfFrZ+jWlcATYkcYaZxJRBf
8mvVQL7kH3BxWXkLcQBT3rMtAMURkeOZlhdKH8wQeHEH6WkPmqypYZlJDVZUdjH3
lX00xKMiRgVY2jB08W9xG04QZDf3NMTBkMUuNVsCAwEAAaOCAoQwggKAMCUGA1Ud
EQQeMByCDSouZWNvdXNlci5uZXSCC2Vjb3VzZXIubmV0MAkGA1UdEwQCMAAwKwYD
VR0fBCQwIjAgoB6gHIYaaHR0cDovL2dwLnN5bWNiLmNvbS9ncC5jcmwwbwYDVR0g
BGgwZjBkBgZngQwBAgEwWjAqBggrBgEFBQcCARYeaHR0cHM6Ly93d3cucmFwaWRz
c2wuY29tL2xlZ2FsMCwGCCsGAQUFBwICMCAMHmh0dHBzOi8vd3d3LnJhcGlkc3Ns
LmNvbS9sZWdhbDAfBgNVHSMEGDAWgBSXwidQnsLJ7AyIMsh8reKmAU/abzAOBgNV
HQ8BAf8EBAMCBaAwHQYDVR0lBBYwFAYIKwYBBQUHAwEGCCsGAQUFBwMCMFcGCCsG
AQUFBwEBBEswSTAfBggrBgEFBQcwAYYTaHR0cDovL2dwLnN5bWNkLmNvbTAmBggr
BgEFBQcwAoYaaHR0cDovL2dwLnN5bWNiLmNvbS9ncC5jcnQwggEDBgorBgEEAdZ5
AgQCBIH0BIHxAO8AdgDd6x0reg1PpiCLga2BaHB+Lo6dAdVciI09EcTNtuy+zAAA
AVkLWYljAAAEAwBHMEUCIBqjefbu1j0kJ5lXW3Rsyw1uXTWKkmDu6hcD45Py3H/J
AiEArLniB1BnNfz/0EzFY4o7TIqSYjx12BmuDuTsLzXO8ycAdQDuS723dc5guuFC
aR+r4Z5mow9+X7By2IMAxHuJeqj9ywAAAVkLWYmjAAAEAwBGMEQCIGdJMC8TGLwG
[... cut ...]
DEBUG    Waiting 1.717647090839548 seconds before connecting.
^C
Aborted!
^CException ignored in: <module 'threading' from '/usr/lib/python3.5/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.5/threading.py", line 1288, in _shutdown
    t.join()
  File "/usr/lib/python3.5/threading.py", line 1054, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.5/threading.py", line 1070, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt

wpietri, 2017-12-09:
Hi, Steve. Thanks for the details. If possible, please attach long dumps like that as files.

This appears to be an issue in `pyasn`. Which version are you using? You can find out doing something like `pip freeze | grep pyasn`. I'm using `pyasn1==0.1.9` which works fine for me. But I see that later versions have an issue that looks like this one:

https://github.com/etingof/pyasn1/issues/112

steveknoefler, 2017-12-09:
Hi,
I changed the version to 0.1.9

root@nd-sv-oh:~# pip3 freeze | grep pyasn
pyasn==1.6.0b1
pyasn1==0.1.9
pyasn1-modules==0.2.1
root@nd-sv-oh:~# pip freeze | grep pyasn
pyasn1==0.1.9
pyasn1-modules==0.2.1

WARNING  Could not find pyasn1 and pyasn1_modules. SSL certificate expiration COULD NOT BE VERIFIED.

[debug.log](https://github.com/wpietri/sucks/files/1545115/debug.log)

steveknoefler, 2017-12-09:
root@nd-sv-oh:/tmp/pyasn1-modules-0.1.5# pip freeze | grep pyasn1
pyasn1==0.3.6
pyasn1-modules==0.1.5

DEBUG    Event triggered: ssl_cert
ERROR    Certificate has expired.
DEBUG    Event triggered: session_end
DEBUG    Event triggered: disconnected
DEBUG     ==== TRANSITION connected -> disconnected
DEBUG    SEND (IMMED): <stream:stream to='ecouser.net' xmlns:stream='http://etherx.jabber.org/streams' xmlns='jabber:client' xml:lang='en' version='1.0'>
DEBUG    Event triggered: socket_error
WARNING  Failed to send b"<stream:stream to='ecouser.net' xmlns:stream='http://etherx.jabber.org/streams' xmlns='jabber:client' xml:lang='en' version='1.0'>"
DEBUG    Event triggered: session_end
DEBUG    Event triggered: socket_error
DEBUG    Event triggered: disconnected
DEBUG     ==== TRANSITION connected -> disconnected
DEBUG    Event triggered: socket_error
ERROR    Socket Error #9: Bad file descriptor
DEBUG    reconnecting...
DEBUG    connecting...
DEBUG    Waiting 2.1675818774356146 seconds before connecting.

steveknoefler, 2017-12-09:
AFTER: 

sudo pip3 uninstall sleekxmpp
sudo pip3 install sleekxmpp==1.3.1

nothing happens to vac :/

DEBUG    Event triggered: ssl_cert
DEBUG    SEND (IMMED): <stream:stream to='ecouser.net' xmlns:stream='http://etherx.jabber.org/streams' xmlns='jabber:client' xml:lang='en' version='1.0'>
DEBUG    RECV: <stream:stream version="1.0" from="ecouser.net" id="203e48766fa206efb7134a3b509f35cc">
DEBUG    RECV: <stream:features xmlns="http://etherx.jabber.org/streams"><auth xmlns="http://jabber.org/features/iq-auth" /><mechanisms xmlns="urn:ietf:params:xml:ns:xmpp-sasl"><mechanism>PLAIN</mechanism></mechanisms></stream:features>
DEBUG    SEND (IMMED): <auth xmlns="urn:ietf:params:xml:ns:xmpp-sasl" mechanism="PLAIN">MjAxNzExMTU1YTBjODEwNjI4YmRjADIwMTcxMTE1NWEwYzgxMDYyOGJkYwAwLzBjMDY3ODEzL094OVpuMU9qaFBsOFVOM1ZYdHI4cno0ZUF2RmZXZHJV</auth>
DEBUG    RECV: <success xmlns="urn:ietf:params:xml:ns:xmpp-sasl" />
DEBUG    Event triggered: auth_success
DEBUG    SEND (IMMED): <stream:stream to='ecouser.net' xmlns:stream='http://etherx.jabber.org/streams' xmlns='jabber:client' xml:lang='en' version='1.0'>
DEBUG    RECV: <stream:stream version="1.0" from="ecouser.net" id="203e48766fa206efb7134a3b509f35cc">
DEBUG    RECV: <stream:features xmlns="http://etherx.jabber.org/streams"><bind xmlns="urn:ietf:params:xml:ns:xmpp-bind" /><session xmlns="urn:ietf:params:xml:ns:xmpp-session" /></stream:features>
DEBUG    Requesting resource: 
DEBUG    SEND (IMMED): <iq type="set" id="c863feb9-746a-42ab-b1ba-7eb8a71e0a51-1"><bind xmlns="urn:ietf:params:xml:ns:xmpp-bind" /></iq>
DEBUG    RECV: <iq type="result" id="c863feb9-746a-42ab-b1ba-7eb8a71e0a51-1"><bind xmlns="urn:ietf:params:xml:ns:xmpp-bind"><jid>201711155a0c810628bdc@ecouser.net/4e2455606f4b3aaad0577f261564d362</jid></bind></iq>
DEBUG    Event triggered: session_bind
INFO     JID set to: 201711155a0c810628bdc@ecouser.net/4e2455606f4b3aaad0577f261564d362
DEBUG    SEND (IMMED): <iq type="set" id="c863feb9-746a-42ab-b1ba-7eb8a71e0a51-2"><session xmlns="urn:ietf:params:xml:ns:xmpp-session" /></iq>
DEBUG    RECV: <iq type="result" id="c863feb9-746a-42ab-b1ba-7eb8a71e0a51-2" />
DEBUG    Established Session
DEBUG    Event triggered: session_start
DEBUG    Finished processing stream features.
DEBUG    Event triggered: stream_negotiated
INFO     CERT: Time until certificate expiration: 8 days, 2:10:29.266315
DEBUG    ----------------- starting session ----------------
DEBUG    event = {}
performing clean command
WARNING  resource property deprecated. Use boundjid.resource
DEBUG    SEND: <iq to="E0001057017700990836@115.ecorobot.net/atom" from="201711155a0c810628bdc@ecouser.net/4e2455606f4b3aaad0577f261564d362" id="c863feb9-746a-42ab-b1ba-7eb8a71e0a51-3" type="set"><query xmlns="com:ctl"><ctl td="Clean"><clean type="auto" speed="standard" /></ctl></query></iq>

wpietri, 2017-12-09:
I'm glad to hear you got past the connection issue.

Perhaps your model uses different commands? Torbjörn was kind enough to document how to try snooping on the exchange between your app and your robot in the [developing.md](https://github.com/wpietri/sucks/blob/master/developing.md#mitm-xmpp-traffic-between-the-android-or-ios-app-and-the-ecovacs-server) file. If you'd like to do that and compare it with the [protocol description](https://github.com/wpietri/sucks/blob/master/protocol.md) for the models we've tried, perhaps you'll see something different.

steveknoefler, 2017-12-10:
hi
its not possible to start the vac :/
I tried to get informations with xmpppeek but nothing happens. Is 5223 the correct port for Germany? 
I checked my robot with an android app for connectivity, is works.

have you an idea? PLEASE 

thanks

wpietri, 2017-12-10:
Hi, Steve. Knowing nothing about your model, it's hard for me to know what the right port number is. If you're familiar with packet captures, you could try getting all the traffic between your app and Ecovacs servers to see what it's doing. You could also try asking Ecovacs for protocol documentation and adding support based on that. Both of these, though, require reasonably advanced development skills. If that's not where you are yet, don't despair; someone else with the same model may come along by and by.

Zenconomy, 2017-12-10:
I'm not sure if this helps, but I've extracted a config file from the Android app that contains country configuration. For Germany:

```
  "DE": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
```

Zenconomy, 2017-12-10:
For xmpppeek you HAVE to use the same server as the app, otherwise traffic will not be intercepted. In your case It's probably msg-eu.ecouser.net:5223

Remember that your router has to override DNS and point msg-eu.ecouser.net to your local machine that runs xmpppeek

Zenconomy, 2017-12-10:
A side note, the Ozmo 930 is a much more advanced bot than what we've been working with so far. Based on what I've found by looking at the App codebase it probably follows the same command structure, but it will be using lot of additional commands to manage things like mapping, geofencing, additional cleaning modes etc.

pilot76103, 2019-02-22:
@wpietri I've been having a detailed conversation with Brian Martin and Jonathan Woodall over in the [Sucks Google Group](https://groups.google.com/forum/#!topic/sucks-users/EiaTYtndkY8). If you would like to review the post I have tried to get some detailed information from the Robot, iOS App and Sucks as separate XMPP traffic. Maybe it can assist in debugging the OZMO 930. Appears to use the same command set.