wpietri/sucks #15: AttributeError: 'module' object has no attribute 'lru_cache'

elbajo, 2017-12-09:
I was trying out the library for the first time and run into a few issues, in the end I figured it out but I thought it might be helpful to other and maybe good to add to the README.md.

After successfully running `sudo pip install -e .`
`sucks login` gives me:

> Traceback (most recent call last):
>   File "/usr/local/bin/sucks", line 11, in <module>
>     load_entry_point('sucks', 'console_scripts', 'sucks')()
>   File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 572, in load_entry_point
>     return get_distribution(dist).load_entry_point(group, name)
>   File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2752, in load_entry_point
>     return ep.load()
>   File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2405, in load
>     return self.resolve()
>   File "/usr/local/lib/python2.7/dist-packages/pkg_resources/__init__.py", line 2411, in resolve
>     module = __import__(self.module_name, fromlist=['__name__'], level=0)
>   File "/opt/ecovacs/sucks/cli.py", line 7, in <module>
>     from pycountry_convert import country_alpha2_to_continent_code
>   File "/usr/local/lib/python2.7/dist-packages/pycountry_convert/__init__.py", line 17, in <module>
>     from .country_mappings import (
>   File "/usr/local/lib/python2.7/dist-packages/pycountry_convert/country_mappings.py", line 15, in <module>
>     @functools.lru_cache()
> AttributeError: 'module' object has no attribute 'lru_cache'

So it seems to be an issue with the pycountry_convert library.

I am kind of a newbie in python and I remember seeing that this was a python3 project.
`python -V`
> Python 2.7.9

So I changed /usr/local/bin/sucks so that it uses python3:
`#!/usr/bin/python3`

Re ran the install commands:
`sudo pip3 install pipenv
sudo pip3 install -e .`

And... tada!

Not sure if I did the right thing here, let me know if not.

Oh and note that my first `sucks charge` call timed out for some reason ("sleekxmpp.exceptions.IqTimeout...") retried and all further calls worked fine.

wpietri, 2017-12-09:
Thanks, @elbajo. I'm going to release this as a proper Python library, which will hopefully prevent people from installing it on incompatible python versions. I appreciate the feedback.

elbajo, 2017-12-10:
Cool, closing this.
Thanks!

closed 2017-12-10 by elbajo