wpietri/sucks #21: support use in China

wpietri, 2017-12-11:
Thanks to @Zenconomy's [decoding of servers by location](https://github.com/wpietri/sucks/blob/master/protocol.md#location), we can now see that the current code doesn't support use in China. The automatic mapping between country and continent could also be improved.

3125589, 2018-03-15:
Is there any plan to support Chinese users?tks!

3125589, 2018-03-26:
This should be the login server address of the Chinese version of the application。
{
  "Chinese": {
    "users": "users.ecouser.net:8000",
    "lg": "lg.ecouser.net:8020",
    "ne": "ne.ecouser.net:8018",
    "dl": "dl.ecouser.net",
    "msg": "msg.ecouser.net:5223",
    "lb": "lb.ecouser.net",
    "AppServer": "app.ecouser.net:8024"
  },
  "TW": {
    "users": "users-as.ecouser.net:8000",
    "lg": "lg-as.ecouser.net:8020",
    "ne": "ne-as.ecouser.net:8018",
    "dl": "dl-as.ecouser.net",
    "msg": "msg-as.ecouser.net:5223",
    "lb": "lb-as.ecouser.net"
  },
  "MY": {
    "users": "users-as.ecouser.net:8000",
    "lg": "lg-as.ecouser.net:8020",
    "ne": "ne-as.ecouser.net:8018",
    "dl": "dl-as.ecouser.net",
    "msg": "msg-as.ecouser.net:5223",
    "lb": "lb-as.ecouser.net"
  },
  "JP": {
    "users": "users-as.ecouser.net:8000",
    "lg": "lg-as.ecouser.net:8020",
    "ne": "ne-as.ecouser.net:8018",
    "dl": "dl-as.ecouser.net",
    "msg": "msg-as.ecouser.net:5223",
    "lb": "lb-as.ecouser.net"
  },
  "SG": {
    "users": "users-as.ecouser.net:8000",
    "lg": "lg-as.ecouser.net:8020",
    "ne": "ne-as.ecouser.net:8018",
    "dl": "dl-as.ecouser.net",
    "msg": "msg-as.ecouser.net:5223",
    "lb": "lb-as.ecouser.net"
  },
  "TH": {
    "users": "users-as.ecouser.net:8000",
    "lg": "lg-as.ecouser.net:8020",
    "ne": "ne-as.ecouser.net:8018",
    "dl": "dl-as.ecouser.net",
    "msg": "msg-as.ecouser.net:5223",
    "lb": "lb-as.ecouser.net"
  },
  "HK": {
    "users": "users-as.ecouser.net:8000",
    "lg": "lg-as.ecouser.net:8020",
    "ne": "ne-as.ecouser.net:8018",
    "dl": "dl-as.ecouser.net",
    "msg": "msg-as.ecouser.net:5223",
    "lb": "lb-as.ecouser.net"
  },
  "IN": {
    "users": "users-as.ecouser.net:8000",
    "lg": "lg-as.ecouser.net:8020",
    "ne": "ne-as.ecouser.net:8018",
    "dl": "dl-as.ecouser.net",
    "msg": "msg-as.ecouser.net:5223",
    "lb": "lb-as.ecouser.net"
  },
  "KR": {
    "users": "users-as.ecouser.net:8000",
    "lg": "lg-as.ecouser.net:8020",
    "ne": "ne-as.ecouser.net:8018",
    "dl": "dl-as.ecouser.net",
    "msg": "msg-as.ecouser.net:5223",
    "lb": "lb-as.ecouser.net"
  },
  "US": {
    "users": "users-na.ecouser.net:8000",
    "lg": "lg-na.ecouser.net:8020",
    "ne": "ne-na.ecouser.net:8018",
    "dl": "dl-na.ecouser.net",
    "msg": "msg-na.ecouser.net:5223",
    "lb": "lb-na.ecouser.net"
  },
  "FR": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "ES": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "UK": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "NO": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "MX": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "DE": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "PT": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "CH": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "AU": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "IT": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "NL": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "SE": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "BE": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "DK": {
    "users": "users-eu.ecouser.net:8000",
    "lg": "lg-eu.ecouser.net:8020",
    "ne": "ne-eu.ecouser.net:8018",
    "dl": "dl-eu.ecouser.net",
    "msg": "msg-eu.ecouser.net:5223",
    "lb": "lb-eu.ecouser.net"
  },
  "default": {
    "users": "users-ww.ecouser.net:8000",
    "lg": "lg-ww.ecouser.net:8020",
    "ne": "ne-ww.ecouser.net:8018",
    "dl": "dl-ww.ecouser.net",
    "msg": "msg-ww.ecouser.net:5223",
    "lb": "lbo.ecouser.net"
  }
}

sjtuross, 2018-10-26:
Is it possible to support China (CN)? China is not in the region list in the Ecovacs app. In China, we need to use this app https://itunes.apple.com/cn/app/%E9%98%BF%E9%87%8C%E6%99%BA%E8%83%BD/id933148709?mt=8 from Alibaba to set up Wifi connection.

wpietri, 2018-10-26:
I believe it is possible to support China. The host names for the various servers use a pattern that's different than the rest of the world. Since Ecovacs doesn't provide test servers or API documentation, this change is best made by somebody in China.

The two parts that need to change for sure are here:

https://github.com/wpietri/sucks/blob/master/sucks/__init__.py#L151

and here:

https://github.com/wpietri/sucks/blob/master/sucks/__init__.py#L169

Ideally, any change would preserve existing behavior while adding China support. But if that's too intimidating, a good place to start would be a patch that makes it work just for China, and we can jointly work on it from there.

jyz0501, 2020-10-08:
Now I provide HAR files of Thor application, hoping to support China.

[ecovacs 20-10-08 13_37_02.zip](https://github.com/wpietri/sucks/files/5345536/ecovacs.20-10-08.13_37_02.zip)

anhtuanng98, 2020-12-07:
Hope our dev can do it :(