wpietri/sucks #27: Error creating config file during login

tucknology, 2017-12-28:
Hi,

I have background in engineering but I'm a bit new to python. I'm running on Mac OS High Sierra, using Python3.

After running `sucks login` and entering the info, I receive the following error:

> Traceback (most recent call last):
>   File "/usr/local/bin/sucks", line 11, in <module>
>     sys.exit(cli())
>   File "/usr/local/lib/python3.6/site-packages/click/core.py", line 722, in __call__
>     return self.main(*args, **kwargs)
>   File "/usr/local/lib/python3.6/site-packages/click/core.py", line 697, in main
>     rv = self.invoke(ctx)
>   File "/usr/local/lib/python3.6/site-packages/click/core.py", line 1092, in invoke
>     rv.append(sub_ctx.command.invoke(sub_ctx))
>   File "/usr/local/lib/python3.6/site-packages/click/core.py", line 895, in invoke
>     return ctx.invoke(self.callback, **ctx.params)
>   File "/usr/local/lib/python3.6/site-packages/click/core.py", line 535, in invoke
>     return callback(*args, **kwargs)
>   File "/usr/local/lib/python3.6/site-packages/sucks/cli.py", line 113, in login
>     write_config(config)
>   File "/usr/local/lib/python3.6/site-packages/sucks/cli.py", line 57, in write_config
>     with open(config_file(), 'w') as fp:
> FileNotFoundError: [Errno 2] No such file or directory: '/Users/aaron/.config/sucks.conf'

My understanding, according to the readme file, is the file should be `.config.sucks.conf` however it is trying to write to `.config/sucks.conf`. I believe the error is at [this line](https://github.com/wpietri/sucks/blob/624d54c77ffb2908a3e0b97a7e981ab510c8c386/sucks/cli.py#L87). Maybe a special case for MacOS must be added? 

Happy to help if I can.

wpietri, 2017-12-28:
Hi, Aaron! Thanks for the report. That is an error in the README; I'll fix that.

The proper location is ~/.config/sucks.conf. But I take it that directory doesn't exist on MacOS. Is it ok if I just create it? Or is there a better place to put per-application config files?

In the meantime, you can get going by just making that directory.

tucknology, 2017-12-28:
Hi wpietri

Thanks for your reply, and for sharing this project with us. This is really great work.

I don't think it will cause any problems to create `~/.config/` directory on MacOS. I think the generally accepted location of storing application-specific configurations in MacOS is in `~/Library/Application Support/Your-App-Name/`, however I'm only familiar with native application (X-Code) and I'm not sure what is standard practice for scripts. Of course, a third option is to create `~/.sucks/`. Quickly scanning my home folder, I see that's how Atom, Dropbox, ssh, etc. operate, but I hate when my home folder gets too cluttered.

I manually created `~/.config` and now I was able to log in, and the script works! Very cool.

torbjornaxelsson, 2017-12-28:
~/Library/Application Support/ is default for native MacOS Applications but many command-line scripts and applications use the ~/.config location. I have 10+ config files there from various scripts and applications. I think it's in the right place and sucks should just create the directory ~/.config if it doen't exist.

wpietri, 2017-12-28:
Fab. Based on this I have released version 0.8.4, which will automatically created the .config directory if it isn't there. That should work fine on Linux and MacOS. (On Windows, we use the APPDATA directory.)

closed 2017-12-28 by wpietri