wpietri/sucks #30: Alexa skill for Deebot N79

jojo3632, 2018-01-18:
Hi,

Can someone help to provide instructions to use this code with Alexa, as an Alexa custom skill, including instructions with speech utterances and Lambda function?

Thanks,
Jordan.

wpietri, 2018-01-19:
Hi, Jordan! I personally can't, and since I'm the only person who reads these new issues, I think you're unlikely to find an answer. I see integrations like that as out of scope for this project, although it seems like a fine idea for some other project to do that using this as a library.

If that is beyond your skills, I'd suggest you find an Alexa-specific forum and post the request there.

closed 2018-01-19 by wpietri

bamminger, 2018-02-26:
I've succesfully managed to run "sucks" in an Alexa lambda function. @jojo3632 if you sill need help, reach out to me on twitter @bamminger

jojo3632, 2018-02-28:
Hi, that's great news. Can you publish the skill or provide me with the zipped files and instructions to do it myself?

bamminger, 2018-02-28:
https://github.com/bamminger/ecovacs-aws

wpietri, 2018-02-28:
Thanks, @bamminger! That's very cool. I'm not a Lambda user, so I don't totally get the issues here. But if there are ways that the library could be better for your use case and still work for the existing ones, I'd happily accept a pull request.

bamminger, 2018-02-28:
There are 2 issues which prevent to use the original version.
1. An AWS specific limitation which seems to make troubles with the underlying Crypto package. The only way we could avoid that is to make it possible to provide the hashed values directly or find an AWS compatible way for the hashing.
2. The [scheduled pinging](https://github.com/wpietri/sucks/blob/bfea9bf2fa4a2ea5187ef1e51312fc455a3dd35a/sucks/__init__.py#L144). Calling the default connect method kept the script running due to the repetitive pinging. What's exactly the purpose of it? In my case the script has to end after I've called the Clean or Charge actions.

EtienneMD, 2018-03-29:
I’ve got a working Alexa skill right now for my Deebot Slim 2. Converted to pycrypto to fix the encryption and I disconnect the XMPP client after I’ve sent a command to end the script. I would have to make a few modifications to the script before I could publish it. Would that be okay with you William? @wpietri

EtienneMD, 2018-04-08:
[https://github.com/EtienneMD/Alexa-Ecovacs]
Just pushed the code this morning. Sorry that took a while, I've got a newborn at home, he is only two weeks old!
Let know if that works on your vac.