wpietri/sucks #32: Getting started using sucks as a library

dixi83, 2018-02-26:
Hello,

I am trying to use use sucks as a library to get som info for my domotica system. I have programming experience but I have not much (almost none) python experience. 

This is the code I have so far for getting the battery state:
```python
from sucks.cli import *

config = read_config()

api = EcoVacsAPI(config['device_id'], config['email'], config['password_hash'], config['country'], config['continent'])
my_vac = api.devices()[0]
vacbot = VacBot(api.uid, api.REALM, api.resource, api.user_access_token, my_vac, config['continent'])
vacbot.connect_and_wait_until_ready()

print(GetBatteryState())
```

But when I execute this code I get a lot of errors:
```
 python3 deebot.py
WARNING:sleekxmpp.basexmpp:fulljid property deprecated. Use boundjid.resource
ERROR:sleekxmpp.basexmpp:day is out of range for month
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1492, in _process
    if not self.__read_xml():
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1564, in __read_xml
    self.__spawn_event(xml)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1632, in __spawn_event
    handler.prerun(stanza_copy)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
    self.run(payload, True)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
    if self.xmpp.start_tls():
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 889, in start_tls
    cert.verify(self._expected_server_name, self._der_cert)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/cert.py", line 133, in verify
    not_before, not_after = extract_dates(raw_cert)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/cert.py", line 111, in extract_dates
    not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
  File "/usr/lib/python3.4/_strptime.py", line 500, in _strptime_datetime
    tt, fraction = _strptime(data_string, format)
  File "/usr/lib/python3.4/_strptime.py", line 465, in _strptime
    datetime_date(year, 1, 1).toordinal() + 1
ValueError: day is out of range for month
ERROR:sleekxmpp.basexmpp:day is out of range for month
```
I installed sucks with "pip install sucks" and I checked if these libraries are installed and "requests-mock, nose, pycountry-convert, pycryptodome, requests, click, sleekxmpp"
I'm proberbly doing something stupid ;) but please help me out

Best Regads, Martijn

wpietri, 2018-02-26:
Hi, Martjin! Does the command-line program work? E.g., can you do `sucks clean 1` from the command-line and have it work? I ask because that will help us figure out whether it's something about your code or something else.

If that doesn't work, then perhaps do `python3 --version` `pip3 list` so we can see if there's some library version issue.

dixi83, 2018-02-27:
Hello wpietri,

Actually not... I was focusing on on my code and I did a "sucks login" (which worked well). So I forgot to test this the output looked like this:
```
# sucks clean 1
ERROR    day is out of range for month
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1492, in _process
    if not self.__read_xml():
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1564, in __read_xml
    self.__spawn_event(xml)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1632, in __spawn_event
    handler.prerun(stanza_copy)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
    self.run(payload, True)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
    if self.xmpp.start_tls():
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 889, in start_tls
    cert.verify(self._expected_server_name, self._der_cert)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/cert.py", line 133, in verify
    not_before, not_after = extract_dates(raw_cert)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/cert.py", line 111, in extract_dates
    not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
  File "/usr/lib/python3.4/_strptime.py", line 500, in _strptime_datetime
    tt, fraction = _strptime(data_string, format)
  File "/usr/lib/python3.4/_strptime.py", line 465, in _strptime
    datetime_date(year, 1, 1).toordinal() + 1
ValueError: day is out of range for month
ERROR    day is out of range for month
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1492, in _process
    if not self.__read_xml():
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1564, in __read_xml
    self.__spawn_event(xml)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1632, in __spawn_event
    handler.prerun(stanza_copy)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
    self.run(payload, True)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
    if self.xmpp.start_tls():
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 889, in start_tls
    cert.verify(self._expected_server_name, self._der_cert)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/cert.py", line 133, in verify
    not_before, not_after = extract_dates(raw_cert)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/cert.py", line 111, in extract_dates
    not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
  File "/usr/lib/python3.4/_strptime.py", line 500, in _strptime_datetime
    tt, fraction = _strptime(data_string, format)
  File "/usr/lib/python3.4/_strptime.py", line 465, in _strptime
    datetime_date(year, 1, 1).toordinal() + 1
ValueError: day is out of range for month
^C
Aborted!
^CException ignored in: <module 'threading' from '/usr/lib/python3.4/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 1294, in _shutdown
    t.join()
  File "/usr/lib/python3.4/threading.py", line 1060, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.4/threading.py", line 1076, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
```

outputs toy requested:
```
# python3 --version
Python 3.4.2
# pip3 list
attrs (17.4.0)
certifi (2018.1.18)
chardet (3.0.4)
click (6.7)
colorama (0.3.2)
coverage (4.5.1)
html5lib (0.999)
idna (2.6)
nose (1.3.7)
pip (1.5.6)
pluggy (0.6.0)
pprintpp (0.3.0)
py (1.5.2)
pyasn1 (0.4.2)
pyasn1-modules (0.2.1)
pycountry (18.2.23)
pycountry-convert (0.7.2)
pycryptodome (3.4.11)
pytest (3.4.1)
pytest-cov (2.5.1)
pytest-mock (1.7.0)
python-dateutil (2.6.0)
repoze.lru (0.7)
requests (2.18.4)
requests-mock (1.4.0)
setuptools (5.5.1)
six (1.11.0)
sleekxmpp (1.3.3)
stringcase (1.2.0)
sucks (0.8.4)
urllib3 (1.22)
virtualenv (1.11.6)
wheel (0.30.0)
```
I am running it on a raspberry pi 3 with Debian Jessie:
```
# uname -a
Linux rpi-1 4.9.35-v7+ #1014 SMP Fri Jun 30 14:47:43 BST 2017 armv7l GNU/Linux
```

dixi83, 2018-02-27:
In my trail on error I tried switching between locales settings with "dpkg-reconfigirue locales" but no luck...

wpietri, 2018-02-27:
For reasons I don't totally understand, this seems to be related to the pyasn1 package. If I have it installed, I get this error. If I don't, it works fine. If for some reason you can't remove that package, you could try rolling back to version 1.3.1 of sleekxmpp, which apparently also fixes this issue: https://github.com/fritzy/SleekXMPP/issues/462

dixi83, 2018-02-27:
Thanks for your reply I managed to remove pyasn1 and i can control (send clean and send charge command) things now. But running  my script still gives errors because of the missing pyasn1 package
```
python3 deebot.py
WARNING:sleekxmpp.basexmpp:fulljid property deprecated. Use boundjid.resource
WARNING:sleekxmpp.xmlstream.cert:Could not find pyasn1 and pyasn1_modules. SSL certificate COULD NOT BE VERIFIED.
WARNING:sleekxmpp.xmlstream.cert:Could not find pyasn1 and pyasn1_modules. SSL certificate expiration COULD NOT BE VERIFIED.
getbatterystate command
```
reverting back to 1.3.1 of sleekxmpp did not change anything... any clues? 

strange is I managed to uninstall pyasn1 with "pip3 uninstall pyasn1" but it is still in the list:
```
# pip3 list
attrs (17.4.0)
certifi (2018.1.18)
chardet (3.0.4)
click (6.7)
colorama (0.3.2)
coverage (4.5.1)
html5lib (0.999)
idna (2.6)
nose (1.3.7)
pip (1.5.6)
pluggy (0.6.0)
pprintpp (0.3.0)
py (1.5.2)
pyasn1 (0.4.2)
pyasn1-modules (0.2.1)
pycountry (18.2.23)
pycountry-convert (0.7.2)
pycryptodome (3.4.11)
pytest (3.4.1)
pytest-cov (2.5.1)
pytest-mock (1.7.0)
python-dateutil (2.6.0)
repoze.lru (0.7)
requests (2.18.4)
requests-mock (1.4.0)
setuptools (5.5.1)
six (1.11.0)
sleekxmpp (1.3.1)
stringcase (1.2.0)
sucks (0.8.4)
urllib3 (1.22)
virtualenv (1.11.6)
wheel (0.30.0)
```

when i reinstall the module with sleekxmpp 1.3.1 installed i get the same errors back: 
```
# python3 deebot.py
WARNING:sleekxmpp.basexmpp:fulljid property deprecated. Use boundjid.resource
ERROR:sleekxmpp.basexmpp:day is out of range for month
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1492, in _process
    if not self.__read_xml():
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1564, in __read_xml
    self.__spawn_event(xml)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1632, in __spawn_event
    handler.prerun(stanza_copy)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
    self.run(payload, True)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
    if self.xmpp.start_tls():
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 889, in start_tls
    cert.verify(self._expected_server_name, self._der_cert)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/cert.py", line 133, in verify
    not_before, not_after = extract_dates(raw_cert)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/cert.py", line 111, in extract_dates
    not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
  File "/usr/lib/python3.4/_strptime.py", line 500, in _strptime_datetime
    tt, fraction = _strptime(data_string, format)
  File "/usr/lib/python3.4/_strptime.py", line 465, in _strptime
    datetime_date(year, 1, 1).toordinal() + 1
ValueError: day is out of range for month
```

dixi83, 2018-03-05:
My trail on error leads to the situation I have now:

for the ValueError I found this solution: https://github.com/fritzy/SleekXMPP/issues/478
using: sleekxmpp 1.3.1, pyasn1 0.4.2 and  pyasn1-modules 0.2.1
I ran into this connection error:
```
# python3 deebot.py
WARNING:sleekxmpp.basexmpp:fulljid property deprecated. Use boundjid.resource
ERROR:sleekxmpp.xmlstream.xmlstream:Connection error.
ERROR:sleekxmpp.basexmpp:<TagSet object at 0x75d22910 tags 0:32:16> not in asn1Spec: <OctetString schema object at 0x75d3c2b0 tagSet <TagSet object at 0x761b2f90 tags 0:0:4> encoding iso-8859-1>
Traceback (most recent call last):
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1481, in _process
    if not self.__read_xml():
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1553, in __read_xml
    self.__spawn_event(xml)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 1621, in __spawn_event
    handler.prerun(stanza_copy)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
    self.run(payload, True)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
    if self.xmpp.start_tls():
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/xmlstream.py", line 882, in start_tls
    cert.verify(self._expected_server_name, self._der_cert)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/cert.py", line 142, in verify
    cert_names = extract_names(raw_cert)
  File "/usr/local/lib/python3.4/dist-packages/sleekxmpp/xmlstream/cert.py", line 73, in extract_names
    asn1Spec=OctetString())[0]
  File "/usr/local/lib/python3.4/dist-packages/pyasn1/codec/ber/decoder.py", line 1318, in __call__
    '%s not in asn1Spec: %r' % (tagSet, asn1Spec)
pyasn1.error.PyAsn1Error: <TagSet object at 0x75d22910 tags 0:32:16> not in asn1Spec: <OctetString schema object at 0x75d3c2b0 tagSet <TagSet object at 0x761b2f90 tags 0:0:4> encoding iso-8859-1>
```

After some googling downgrading versions of the used packages:
using: sleekxmpp 1.3.1, pyasn1 0.3.7 and  pyasn1-modules 0.1.5
I only get into a Warning loop and no actual state of the battery :(
```
# python3 deebot.py
WARNING:sleekxmpp.basexmpp:fulljid property deprecated. Use boundjid.resource
getbatterystate command
WARNING:sleekxmpp.basexmpp:resource property deprecated. Use boundjid.resource
WARNING:sleekxmpp.basexmpp:resource property deprecated. Use boundjid.resource
WARNING:sleekxmpp.basexmpp:resource property deprecated. Use boundjid.resource
WARNING:sleekxmpp.basexmpp:resource property deprecated. Use boundjid.resource
WARNING:sleekxmpp.basexmpp:resource property deprecated. Use boundjid.resource
WARNING:sleekxmpp.basexmpp:resource property deprecated. Use boundjid.resource
^CException ignored in: <module 'threading' from '/usr/lib/python3.4/threading.py'>
Traceback (most recent call last):
  File "/usr/lib/python3.4/threading.py", line 1294, in _shutdown
    t.join()
  File "/usr/lib/python3.4/threading.py", line 1060, in join
    self._wait_for_tstate_lock()
  File "/usr/lib/python3.4/threading.py", line 1076, in _wait_for_tstate_lock
    elif lock.acquire(block, timeout):
KeyboardInterrupt
```
Any suggestions what is left to try or debug for me? 
B.T.W. the `sucks clean 1`  and `sucks charge` commands still work ;)

dixi83, 2018-03-05:
OK after I was writing my previous post I decided to go back to my code.
I can confirm this script works a kind of for me now:
```
from sucks.cli import *

config = read_config()

api = EcoVacsAPI(config['device_id'], config['email'], config['password_hash'], config['country'], config['continent'])
my_vac = api.devices()[0]
vacbot = VacBot(api.uid, api.REALM, api.resource, api.user_access_token, my_vac, config['continent'])
vacbot.connect_and_wait_until_ready()

vacbot.run(Clean())  # start cleaning
time.sleep(30)      # clean for 1/2 minute
vacbot.run(Charge()) # return to the charger
```
it does not return to the charger after 30 seconds...

I tried getting the battery state again. but no luck...
```
from sucks.cli import *

config = read_config()

api = EcoVacsAPI(config['device_id'], config['email'], config['password_hash'], config['country'], config['continent'])
my_vac = api.devices()[0]
vacbot = VacBot(api.uid, api.REALM, api.resource, api.user_access_token, my_vac, config['continent'])
vacbot.connect_and_wait_until_ready()

test = vacbot.run(GetBatteryState())
print(test)
```
It seems to me that the script is "hanging" after the first "run" line

snipem, 2018-03-07:
I have the same problem as the original poster.

This is on my Linux server. Works on my Mac!

(I removed IDs and certificates)
```
~  $ sucks --debug stop
DEBUG    will run [<sucks.cli.CliAction object at 0x7f5d2f177470>]
DEBUG    Setting up EcoVacsAPI
DEBUG    calling main api user/login with (('account ...
DEBUG    Starting new HTTPS connection (1): eco-de-api.ecovacs.com
DEBUG    https://eco-de-api.ecovacs.com:443 "GET / ......
DEBUG    got {'code': '0000', 'msg': '操作成功', 'time': 1520450923676, 'data': {'uid': '......', 'username': None, 'email': '......', 'country': 'de', 'accessToken': '...'}}
DEBUG    calling main api user/getAuthCode with (('uid', ... ('accessToken',... 
DEBUG    Starting new HTTPS connection (1): eco-de-api.ecovacs.com
DEBUG    https://eco-de-api.ecovacs.com:443 "GET ... HTTP/1.1" 200 None
DEBUG    got {'code': '0000', 'msg': '操作成功', 'time': 1520450923803, 'data': {'authCode':... 
DEBUG    calling user api loginByItToken with {'country': 'DE', 'resource': '9f2e6e14', 'realm': 'ecouser.net', 'userId': ... 'token':... 
DEBUG    Starting new HTTPS connection (1): users-eu.ecouser.net
DEBUG    https://users-eu.ecouser.net:8000 "POST /user.do HTTP/1.1" 200 129
DEBUG    got {'todo': 'result', 'result': 'ok', 'userId': ... 'resource': '9f2e6e14', 'token':... 
DEBUG    EcoVacsAPI connection complete
DEBUG    calling user api GetDeviceList with {'userid': ... 'auth': {'with': 'users', 'userid': ... 'realm': 'ecouser.net', 'token': ... 'resource': '9f2e6e14'}}
DEBUG    Starting new HTTPS connection (1): users-eu.ecouser.net
DEBUG    https://users-eu.ecouser.net:8000 "POST /user.do HTTP/1.1" 200 164
DEBUG    got {'todo': 'result', 'result': 'ok', 'devices': [{'did': ... 'name': ... 'class': '123', 'resource': 'atom', 'nick': None, 'company': 'eco'}]}
DEBUG    Loaded Plugin: RFC 6120: Stream Feature: STARTTLS
DEBUG    Loaded Plugin: RFC 6120: Stream Feature: Resource Binding
DEBUG    Loaded Plugin: RFC 3920: Stream Feature: Start Session
DEBUG    Loaded Plugin: RFC 6121: Stream Feature: Roster Versioning
DEBUG    Loaded Plugin: RFC 6121: Stream Feature: Subscription Pre-Approval
DEBUG    Loaded Plugin: RFC 6120: Stream Feature: SASL
WARNING  fulljid property deprecated. Use boundjid.resource
DEBUG    Waiting 1.763896041315836 seconds before connecting.
DEBUG    DNS: Querying msg-eu.ecouser.net for AAAA records.
DEBUG    DNS: Error retreiving AAAA address info for msg-eu.ecouser.net.
DEBUG    DNS: Querying msg-eu.ecouser.net for A records.
DEBUG    Connecting to 47.91.67.63:5223
DEBUG    Event triggered: connected
DEBUG     ==== TRANSITION disconnected -> connected
DEBUG    Starting HANDLER THREAD
DEBUG    Loading event runner
DEBUG    SEND (IMMED): <stream:stream to='ecouser.net' xmlns:stream='http://etherx.jabber.org/streams' xmlns='jabber:client' xml:lang='en' version='1.0'>
DEBUG    RECV: <stream:stream version="1.0" id="45c63990a9fd5ca2c3268d75319bf37b" from="ecouser.net">
DEBUG    RECV: <stream:features xmlns="http://etherx.jabber.org/streams"><auth xmlns="http://jabber.org/features/iq-auth" /><starttls xmlns="urn:ietf:params:xml:ns:xmpp-tls"><required /></starttls><mechanisms xmlns="urn:ietf:params:xml:ns:xmpp-sasl"><mechanism>PLAIN</mechanism></mechanisms></stream:features>
DEBUG    SEND (IMMED): <starttls xmlns="urn:ietf:params:xml:ns:xmpp-tls"><required /></starttls>
DEBUG    RECV: <proceed xmlns="urn:ietf:params:xml:ns:xmpp-tls" />
DEBUG    Starting TLS
INFO     Negotiating TLS
INFO     Using SSL version: TLSv1
DEBUG    CERT: -----BEGIN CERTIFICATE-----
... 
-----END CERTIFICATE-----

DEBUG    Event triggered: ssl_cert
ERROR    day is out of range for month
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1492, in _process
    if not self.__read_xml():
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1564, in __read_xml
    self.__spawn_event(xml)
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1632, in __spawn_event
    handler.prerun(stanza_copy)
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
    self.run(payload, True)
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/lib/python3.6/site-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
    if self.xmpp.start_tls():
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 889, in start_tls
[... 195 lines cut ...]
six (1.11.0)
sleekxmpp (1.3.3)
slip (0.6.4)
slip.dbus (0.6.4)
SQLAlchemy (1.2.4)
stringcase (1.2.0)
sucks (0.8.4)
systemd-python (233)
texttable (0.8.8)
TwitterAPI (2.4.8)
uritemplate (3.0.0)
urllib3 (1.21.1)
w3lib (1.19.0)
websocket-client (0.40.0)
websockets (4.0.1)
Werkzeug (0.12.1)
wheel (0.30.0)
youtube-dl (2017.10.7)
youtube-upload (0.8.0)
```

```
~  $ uname -a
Linux server 4.14.12-1-ARCH #1 SMP PREEMPT Fri Jan 5 18:19:34 UTC 2018 x86_64 GNU/Linux
```

dixi83, 2018-03-08:
hello snipem,

I managed to solve the most of these problems.  I also confirmed the command line function is working with sleekxmpp 1.3.1. See read my previous posts.

downgrade to these versions:
sleekxmpp 1.3.1, pyasn1 0.3.7 and pyasn1-modules 0.1.5
by using `pip3 install sleekxmpp==1.3.1` and for the other two packages the same way.

snipem, 2018-03-08:
Yep, I made it working with these changes: https://github.com/snipem/sucks/commit/b95bbaf645c37aaf6de72d4321323501e8b7229f

I basically fixed the versions that were stated in this issue.

closed 2018-07-17 by wpietri

SwapFaceL, 2018-10-09:
When I start the script with the downgrades the deebot starts as well but the script remains in an endless loop. Any idea how to stop it after sending the command?

guillebot, 2018-11-23:
@SwapFaceL I'm getting the same results. It works fine but the python script never exits. Have you solved this? I think some thread with the VacBot object must be still running and I don't know how to stop it. 

thanks

SwapFaceL, 2018-11-23:
@guillebot Yes I've solved this problem. I am not at home, if I do not forget, I can post you the solution later or maybe someone else who is faster ;)

guillebot, 2018-11-23:
Yet another self answer:

```vacbot.disconnect(wait=True)```

did the trick.

thanks anyway

Neustradamus, 2018-11-30:
Have you tested with SleekXMPP 1.3.2?

guillebot, 2018-11-30:
Not yet.

I just updated and it went straight throug 1.3.3.

It's that ok? Or sucks needs 1.3.2?

thanks!

guillebot, 2018-11-30:
No luck it seems.

Updated to 1.3.3

```
openhab:~ # pip list |grep xm
sleekxmpp         1.3.3
```

Same error:

```Device ID: E0000693817701104736
WARNING:sleekxmpp.basexmpp:fulljid property deprecated. Use boundjid.resource
WARNING:sleekxmpp.xmlstream.cert:Could not find pyasn1 and pyasn1_modules. SSL certificate COULD NOT BE VERIFIED.
WARNING:sleekxmpp.xmlstream.cert:Could not find pyasn1 and pyasn1_modules. SSL certificate expiration COULD NOT BE VERIFIED.
Battery level: 100.0
Vacuum status:charging
Clean status:stop
Charge Status: charging
Lifespan: {"type": "filter", "lifespan": 0.44}
tipo: filter
valor: 44.0
Lifespan: {"type": "main_brush", "lifespan": 0.69}
tipo: main_brush
valor: 69.0
Lifespan: {"type": "side_brush", "lifespan": 0.39}
tipo: side_brush
valor: 39.0
Subscribe topic: ecovacs/E0000693817701104736/command
Lifespan: {"type": "filter", "lifespan": 0.44}
tipo: filter
valor: 44.0
Status: idle
Vacuum status:idle
Clean status:stop
Charge Status: idle
Status: auto
Vacuum status:auto
Clean status:auto
Charge Status: idle
ERROR:sleekxmpp.xmlstream.xmlstream:Error processing stream handler: general
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/lib/python3.6/site-packages/sucks/__init__.py", line 458, in _handle_ctl
    s(as_dict)
  File "/usr/lib/python3.6/site-packages/sucks/__init__.py", line 289, in _handle_ctl
    getattr(self, method)(ctl)
  File "/usr/lib/python3.6/site-packages/sucks/__init__.py", line 292, in _handle_error
    error = event['error']
KeyError: 'error'
ERROR:sleekxmpp.stanza.rootstanza:Error handling {jabber:client}iq stanza
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/lib/python3.6/site-packages/sucks/__init__.py", line 458, in _handle_ctl
    s(as_dict)
  File "/usr/lib/python3.6/site-packages/sucks/__init__.py", line 289, in _handle_ctl
    getattr(self, method)(ctl)
  File "/usr/lib/python3.6/site-packages/sucks/__init__.py", line 292, in _handle_error
    error = event['error']
KeyError: 'error'
ERROR:sleekxmpp.basexmpp:'error'
Traceback (most recent call last):
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/usr/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/lib/python3.6/site-packages/sucks/__init__.py", line 458, in _handle_ctl
    s(as_dict)
  File "/usr/lib/python3.6/site-packages/sucks/__init__.py", line 289, in _handle_ctl
    getattr(self, method)(ctl)
  File "/usr/lib/python3.6/site-packages/sucks/__init__.py", line 292, in _handle_error
    error = event['error']
KeyError: 'error'
```

Neustradamus, 2019-04-08:
@dixi83 @wpietri @snipem @SwapFaceL @guillebot: Have you tested the "master"?
- https://github.com/fritzy/SleekXMPP/commits/develop