wpietri/sucks #33: GetDeviceList failed

eracknaphobia, 2018-04-05:
None of the calls seem to work, with the exception of login. I repeatedly get:
`ERROR    call to GetDeviceList failed with {'result': 'fail', 'error': 'auth error', 'todo': 'result', 'errno': 1004}`
 
Ideas?

torbjornaxelsson, 2018-04-08:
Could be a deliberate API change on the server side, or a server error... Which vacuum models do you guys have? Has Sucks previously worked for you? Is the official app working?

eracknaphobia, 2018-04-08:
Deebot N79s

tripzero, 2018-04-08:
Same. N79s.  First time trying sucks.

vietquocnguyen, 2018-04-10:
Same here, Deebot N79s. The official app works. I'm guessing they changed something on the server side?

tripzero, 2018-04-10:
is it particular to the N79s?  I wonder if the "s" is treated differently on the server side than the N79?

ttrushin, 2018-04-11:
Same error here. First time using sucks. Deebot N79.

tripzero, 2018-04-11:
@ttrushin N79? Not the "S" variant?

ttrushin, 2018-04-11:
@tripzero It's the N79, not the "S" variant, yep

vendorstring, 2018-04-12:
R98 same error, official app works

gdgeist, 2018-04-12:
N79S and yes worked prior (like a charm!).  Thanks for the code here!  Was using this to integrate with Openhab.

thimslugga, 2018-05-17:
I'm experiencing this error, possible they changed something on their end?

joey52685, 2018-06-02:
Same problem.  Anybody have any luck figuring out the issue?  Was this confirmed as an API change?

tripzero, 2018-06-02:
A pull request was submitted with a fix. I've been using that fork for now.

thimslugga, 2018-06-04:
I've applied the fix in the pr to my own fork but running into some other errors now. The commands work but need to do some additional debugging when I get a chance

trhr, 2018-06-09:
#34 works for me.

thimslugga, 2018-06-09:
@trhr did you figure out a workaround for the errors after each command?

trhr, 2018-06-10:
@thimslugga I didn't debug any errors after making the changes in #34 . I'm running Python3.5.3 on Debian Stretch, and the commands work as intended, so I'm satisfied.

salanki, 2018-07-22:
#34 fixed it for me as well

wpietri, 2018-07-22:
As I mentioned on the pull request, it's not clear to me why some people need this. If people who have this problem could say more about their setup so we can figure out the difference, I'd appreciate it. For example, I'd love to know things like:

    Operating system version
    Python version
    Vacuum model
    Country code and continent code
    How many Ecovacs devices you have

I use this daily and have never seen the error, so I'll also happily accept session captures or any other data people think might be helpful.

derVicthor, 2018-07-23:
Linux  4.14.50-v7+
Python 3.5.3
Deebot Slim2
DE / EU
only one

For me sucks never worked without that fix. With the fix, I am getting a timeout from the XMPP, but the vacuum is cleaning.

eracknaphobia, 2018-07-23:
- Raspbian GNU/Linux 9.4 (stretch)
- Python 3.6.5
- N79s
- US / NA
- 1

sleepyarmistice, 2018-07-24:
MacOS 10.13.5
Python 3.6.6
N79S
US/NA
1

elijahnyp, 2018-07-24:
#34 fixed it for me.  I noticed that the auth information received appears to be good for a bit.  Perhaps the code should be modified so it stores the new uid and token and only goes through the full login routine if it gets a 1004 error back on the user_api.  Maybe I'll have a chance to implement this later after I've done some more testing.

DJ1975-SE, 2018-08-05:
Linux 64bit v 4.17.11
Python 3.7.0b1+
M81 Pro
DE/EU
1

Android app worked. New device, so I have never used sucks previously. 

I got the 1004 as others above. Applied https://github.com/wpietri/sucks/pull/34 and it started working (didn't even have to regenerate config file). 

If a tcpdump or xmppeek is of any use, let me know and I'll get it setup. Thanks for a great project.

wpietri, 2018-08-05:
Thanks, @DJ1975-SE, I'd love that. It's HTTPS, so I've been using mitmproxy to get the decoded exchange. It would be great to get the XMPP stuff as well just to make sure that isn't any different.

DJ1975-SE, 2018-08-07:
There are mitmproxy-dumps under https://github.com/DJ1975-SE/scratchplace/ from the current 0.8.4 version as well as with the patch applied.

Additionally, the config files created are also there.

It's all unredacted but the credentials aren't valid any longer.

Hope this helps! I'll try to capture some other things now, I believe someone looked for the find vacuum command. If the dumpfiles can be improved or if there are any other dumps I can provide drop me a line.

TonyFeestneus, 2018-08-30:
This has to do with the wrong User ID, which can be seen if you use --debug with the command. 
Like: sudo sucks --debug clean 10

Before the error, it returns a shorter userID, if you hardecode that in to __init__.py, the login is succesful after that

jcconnell, 2018-09-01:
@TonyFeestneus are you referring to `self.uid` on line 121 in `__init__.py`?

I ran the command with `--debug` as you suggested and I see what I think is the shorter username in the output (next to my email).

TonyFeestneus, 2018-09-01:
Yes. I added a line I think on line 129 which makes it work.

`self.uid = login_response['userId']`

TonyFeestneus, 2018-09-04:
I have line 121 original:

`self.uid = login_info['uid']`

and inserted a line at 129:
`self.uid = login_response['userId']`

I think the first one is needed to get the authcode, and setting it at 129 makes sure the correct one is used in function def devices.

Otherwise hardocde your userid (in single parenthesis) in this part (2x where you see self.uid):
`def devices(self):
        _LOGGER.debug("got hier 1")
        devices = self.__call_user_api('GetDeviceList', {
            'userid': self.uid,
            'auth': {
                'with': 'users',
                'userid': self.uid,
                'realm': EcoVacsAPI.REALM,
                'token': self.user_access_token,
                'resource': self.resource
            }
        })['devices']
        return devices`

jcconnell, 2018-09-05:
For anyone who finds themselves here, this [patch](https://github.com/wpietri/sucks/pull/34/commits/8f3bf0ac60164450edd09c5f97bdd4bc06f0a62f) works to get the vacuum started but there are all kinds of errors once it's going. 

Example errors:

```
sleekxmpp.basexmpp WARNING  fulljid property deprecated. Use boundjid.resource
sleekxmpp.xmlstream.cert WARNING  Could not find pyasn1 and pyasn1_modules. SSL certificate COULD NOT BE VERIFIED.
sleekxmpp.xmlstream.cert WARNING  Could not find pyasn1 and pyasn1_modules. SSL certificate expiration COULD NOT BE VERIFIED.

sleekxmpp.basexmpp WARNING  resource property deprecated. Use boundjid.resource

sleekxmpp.xmlstream.xmlstream ERROR    Error processing stream handler: general
```

I checked some of these errors and what others had done to solve them and discovered that sucks made some changes to it's version requirements. Versions of [sucks required packages](https://github.com/snipem/sucks/blob/master/setup.py) installed in my Home Assistant virtual environment were:
sucks==0.9.1
sleekxmpp==1.3.3 (vs 1.3.1 requirement)
stringcase==1.2.0
pyasn1 missing
pyasn1-modules missing

So I installed pyasn1 and started seeing a new error:
```
sleekxmpp.basexmpp ERROR    day is out of range for month
Traceback (most recent call last):
  File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1492, in _process
    if not self.__read_xml():
  File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1564, in __read_xml
    self.__spawn_event(xml)
  File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1632, in __spawn_event
    handler.prerun(stanza_copy)
  File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
    self.run(payload, True)
  File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
    if self.xmpp.start_tls():
  File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 889, in start_tls
    cert.verify(self._expected_server_name, self._der_cert)
  File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/cert.py", line 133, in verify
    not_before, not_after = extract_dates(raw_cert)
  File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/cert.py", line 111, in extract_dates
    not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
  File "/usr/lib/python3.6/_strptime.py", line 565, in _strptime_datetime
    tt, fraction = _strptime(data_string, format)
  File "/usr/lib/python3.6/_strptime.py", line 528, in _strptime
    datetime_date(year, 1, 1).toordinal() + 1
ValueError: day is out of range for month
```

Determined, then I rolled back `sleekxmpp==1.3.3` to `sleekxmpp==1.3.1` as required in the sucks `setup.py`.  Now the vacuum works via the sucks library in the Home Assistant virtual environment and the only errors I'm receiving look like:
`sleekxmpp.basexmpp WARNING  resource property deprecated. Use boundjid.resource`

The vacuum displays in the Home Assistant frontend as `vacuum.none`. All features seem to work now.

OverloadUT, 2018-09-16:
Hey @jcconnell - Why are you using the `snipem/sucks` fork to determine which packages should be installed? I see that snipem changed the dependencies in his fork but I am not sure why. pyasn1, for example, should not be a `sucks` requirement because `sucks` never uses it directly. If it's used by sleekxmpp, it would be marked as a dependency in that package.

jcconnell, 2018-09-17:
@OverloadUT This thread and the final post there are what lead me to use the dependencies from his fork: https://github.com/wpietri/sucks/issues/32. I wish I had a better reason but it was just the result of trial and error.

closed 2018-09-23 by wpietri

DemiMelfice, 2018-11-05:
@jcconnell Could I ask how you downgraded sleekxmpp from 1.3.3 to 1.3.1?

My Home Assistant is running from a Docker on an unRaid setup, i've checked and can see that I have sleekxmpp 1.3.3 installed within the Docker however I'm not sure howto remove this and install the 1.3.1 version.

I've also made the changes to the cert.py file as suggested [fritzy/SleekXMPP#478](https://github.com/fritzy/SleekXMPP/issues/478) however I believe the roll back to 1.3.1 should resolve this for me.