wpietri/sucks #35: Help integrating with homebridge?

raygan, 2018-04-23:
I'm trying to use this to integrate a Deebot N79 with HomeKit via [homebridge](https://github.com/nfarina/homebridge), and I've encountered a bit of a problem. Apologies if this is obvious, but I'm not a programmer, just a tinkerer.

With homebridge I can use plugins like [cmdswitch2](https://github.com/luisiam/homebridge-cmdswitch2) to run command line commands as switches via HomeKit. cmdSwitch2 lets me define an "on command" to run when the switch is turned on, an "off command" to run when the switch is turned off, and a "state command" that can be run periodically to check the state of the device and tell if it's currently on or off. Usually this is used for something like a wakeonlan command (on), a sleep command (off), and something like a ping to tell if the target device is on (status).

I'd like to set a long or open ended "clean" command as an "on command" but be able to interrupt it with a "charge" or "stop" command at any time, rather than having the script run for the entire cleaning cycle. It would also be nice if there were a command that would output the current state of the vacuum. Is something like that possible?

raygan, 2018-04-24:
Thanks, that's very helpful. Any chance you could share your state_command script? I don't know Python and would love an example to work from.

EtienneMD, 2018-04-24:
Yes of course. I’ll publish sometime this week. I will keep you posted.

tucknology, 2018-04-24:
@EtienneMD I am also interested to see how you’ve set it up. Thanks in advance.

EtienneMD, 2018-04-29:
https://github.com/EtienneMD/Cmdswitch2-Ecovacs

raygan, 2018-05-01:
Thanks for your script! This worked pretty well. I hope Apple eventually adds a vacuum device category to homekit so we can have a more full featured homebridge plugin but this will work for now.

closed 2018-05-01 by raygan