wpietri/sucks #36: Errors when run via homebridge

raygan, 2018-05-04:
When I run sucks from the terminal it works, but when I run it via the CmdSwitch2 plugin in Homebridge it sometimes fails with the following error:

```
[5/3/2018, 6:37:21 PM] [CMD Switch] Vacuum is off.
[5/3/2018, 6:37:22 PM] [CMD Switch] Failed to turn on Vacuum
[5/3/2018, 6:37:22 PM] [CMD Switch] Traceback (most recent call last):
  File "/usr/local/bin/sucks", line 11, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.6/site-packages/click/core.py", line 676, in main
    _verify_python3_env()
  File "/usr/local/lib/python3.6/site-packages/click/_unicodefun.py", line 118, in _verify_python3_env
    'for mitigation steps.' + extra)
RuntimeError: Click will abort further execution because Python 3 was configured to use ASCII as encoding for the environment.  Consult http://click.pocoo.org/python3/for mitigation steps.

This system lists a couple of UTF-8 supporting locales that
you can pick from.  The following suitable locales where
discovered: af_ZA.UTF-8, am_ET.UTF-8, be_BY.UTF-8, bg_BG.UTF-8, ca_ES.UTF-8, cs_CZ.UTF-8, da_DK.UTF-8, de_AT.UTF-8, de_CH.UTF-8, de_DE.UTF-8, el_GR.UTF-8, en_AU.UTF-8, en_CA.UTF-8, en_GB.UTF-8, en_IE.UTF-8, en_NZ.UTF-8, en_US.UTF-8, es_ES.UTF-8, et_EE.UTF-8, eu_ES.UTF-8, fi_FI.UTF-8, fr_BE.UTF-8, fr_CA.UTF-8, fr_CH.UTF-8, fr_FR.UTF-8, he_IL.UTF-8, hr_HR.UTF-8, hu_HU.UTF-8, hy_AM.UTF-8, is_IS.UTF-8, it_CH.UTF-8, it_IT.UTF-8, ja_JP.UTF-8, kk_KZ.UTF-8, ko_KR.UTF-8, lt_LT.UTF-8, nl_BE.UTF-8, nl_NL.UTF-8, no_NO.UTF-8, pl_PL.UTF-8, pt_BR.UTF-8, pt_PT.UTF-8, ro_RO.UTF-8, ru_RU.UTF-8, sk_SK.UTF-8, sl_SI.UTF-8, sr_YU.UTF-8, sv_SE.UTF-8, tr_TR.UTF-8, uk_UA.UTF-8, zh_CN.UTF-8, zh_HK.UTF-8, zh_TW.UTF-8
```

Any idea what might be happening here?

Here is the relevant section of my homebridge config.json:

```
{
	"platform": "cmdSwitch2",
	"name": "CMD Switch",
	"switches": [{
     	"name" : "Vacuum",
     	"on_cmd": "sucks clean 120",
     	"off_cmd": "sucks charge",
     	"state_cmd": "python3 ~/.homebridge/vacstate.py",
     	"polling": true,
     	"interval": 120,
     	"timeout": 20
                 }]
},
```

When I flip the switch in the Home app or ask Siri to turn on the vacuum, it's attempting to run `sucks clean 120` but getting the above error.

raygan, 2018-05-04:
I believe I'm already set to UTF-8. Here's what I get when run `locale`:

```
mini:~ raygan$ locale
LANG="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_CTYPE="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_ALL=
```

EtienneMD, 2018-05-05:
Your issue is documented here:
http://click.pocoo.org/5/python3/

raygan, 2018-05-11:
Thanks. I still don't really know why homebridge was restricting it to ascii, but adding `export LC_ALL=en_US.utf-8 && export LANG=en_US.utf-8 && ` before each of my `sucks` commands in config.json seems to have fixed the issue.

closed 2018-05-11 by raygan