wpietri/sucks #37: D900 unavailable

Bolli84, 2018-05-07:
My D900 is with the patched sucks version not working. 
At first, there was the "GetDeviceList"-Failure, but I changed the code and now,the following issue appears:

sleekxmpp.exceptions.IqError:` <iq id="80a2b069-2a65-41fe-8e54-050373ed4197-4" from="8d2fdd68-5eaf-4e59-9cb8-67776a392875@ls1ok3.ecorobot.net/atom" type="error" to="cedrblwf12886d5c@ecouser.net/09a8224604a2b597bf75b4c3d3c26295"><error code="404" type="wait"><recipient-unavailable xmlns="urn:ietf:params:xml:ns:xmpp-stanzas" /></error></iq>

The used ID's from above don't match the ID's which are used in my APP but the login was correctly registered.

Checking it with the APP, it works well. 

Any Idea

wpietri, 2018-05-22:
I don't think anybody has added support for the D900 yet. If you would like add it, the first step is to snoop the protocol between your app and your robot. Details here:

https://github.com/wpietri/sucks/blob/master/developing.md

joey52685, 2018-06-02:
Same issue here.  I fixed the GetDeviceList issue and now running into the sleekxmpp.exceptions.IqError:` with the Deebot 900.

I'll see if I can find time during the week to run through the xmpppeek process.

rgabrielson11, 2018-09-25:
Would like to assist with getting the 900 series functioning.  Whats the status/what is needed?  My programming expertise is minimal, but should be able to collect data if needed from my unit.

kquinsland, 2018-10-05:
I am running into issues getting the 901 working with HomeAssistant.  if the 900 series turns out to do something new with the protocol and is not supported, then i'll see what i can do to sniff the traffic and help.

is this the same `IQError` you get @Joey40 

```
INFO (event_thread_0) [sleekxmpp.xmlstream.xmlstream] Waiting for </stream:stream> from server
ERROR (read_thread) [sleekxmpp.xmlstream.xmlstream] Error reading from XML stream.
ERROR (read_thread) [sleekxmpp.xmlstream.xmlstream] Error reading from XML stream.
WARNING (event_thread_0) [sleekxmpp.xmlstream.handler.waiter] Timed out waiting for IqWait_UID6
WARNING (event_thread_0) [sucks] Ping did not reach VacBot. Will retry.
WARNING (SyncWorker_3) [sleekxmpp.xmlstream.handler.waiter] Timed out waiting for IqWait_UID5
ERROR (MainThread) [homeassistant.core] Error doing job: Task exception was never retrieved
Traceback (most recent call last):
  File "/usr/src/app/homeassistant/helpers/service.py", line 224, in _handle_service_platform_call
    await getattr(entity, func)(**data)
  File "/usr/src/app/homeassistant/components/vacuum/__init__.py", line 387, in async_turn_on
    partial(self.turn_on, **kwargs))
  File "/usr/local/lib/python3.6/concurrent/futures/thread.py", line 56, in run
    result = self.fn(*self.args, **self.kwargs)
  File "/usr/src/app/homeassistant/components/vacuum/ecovacs.py", line 151, in turn_on
    self.device.run(Clean())
  File "/usr/local/lib/python3.6/site-packages/sucks/__init__.py", line 416, in run
    self.send_command(action.to_xml())
  File "/usr/local/lib/python3.6/site-packages/sucks/__init__.py", line 413, in send_command
    self.xmpp.send_command(xml, self._vacuum_address())
  File "/usr/local/lib/python3.6/site-packages/sucks/__init__.py", line 482, in send_command
    c.send()
  File "/usr/local/lib/python3.6/site-packages/sleekxmpp/stanza/iq.py", line 233, in send
    raise IqTimeout(self)
sleekxmpp.exceptions.IqTimeout: <iq id="UID5" to="UID4@ls1ok3.ecorobot.net/atom" from="UID_1@ecouser.net/UID_2" type="set"><query xmlns="com:ctl"><ctl td="Clean"><clean type="auto" speed="standard" /></ctl></query></iq>
WARNING (event_thread_0) [sleekxmpp.xmlstream.handler.waiter] Timed out waiting for IqWait_UID3
WARNING (event_thread_0) [sucks] Ping did not reach VacBot. Will retry.

things to note `ls1ok3.ecorobot.net` is not a valid domain; i don't know if this is related
```

rgabrielson11, 2018-10-05:
Yep that’s what I’m seeing too. 

---sent from my iPhone

ecpunk, 2019-01-28:
@wpietri, the 900 uses the same msg-na.ecouser.net (47.88.66.164) for the HTTPS as it does for 5223. Below is a capture directly from the app to the internet.

[image]

When I change DNS and try to capture via xmpppeek.py, no dice. The machine with xmpppeek isn't listening on 443. 

[image]

On other robots how are you accomplishing the capture? Is the https going to the same URL as the xmpp traffic?

wpietri, 2019-01-28:
At this point I don't remember, but I was using [mitmproxy](https://mitmproxy.org/) to capture the HTTPS traffic, so that sounds like a good idea to me.

mariuskiessling, 2019-02-07:
@bmartin5692 I have just checked out your fork and it works great with my DEBOOT 901! Just out of interest, how did you name your areas? I saw that you also added the spotarea command to the CLI but I can't figure out how names are assigned to areas based on your commits.

bmartin5692, 2019-02-08:
@mariuskiessling  Glad it works for you, it is nice to see confirmation.  

**RE:** SpotAreas - I'll make sure to document this as well, but for the below to work you'll need to get my latest commit in the 901 branch.  

### SpotAreas
For the CLI - the `spotclean` command takes a csv of ints and minutes to clean - ex `spotclean 0,1 60`
For the Library you could use `vacbot.run(SpotArea('start', '0,1'))`

"0,1" is a list of mapIDs the bot should clean.  Each of these corresponds to a room or area the bot mapped. In the app, these are what show the letters over rooms mapID (0) == room ("A"), (1) == "B", etc.

You'll need to use the library, but if you set --debug for sucks you can get your SpotArea MapSet with:
`vacbot.run(VacBotCommand("GetMapSet", {"tp":"sa"}))`
You'll see in DEBUG something like:
```
sucks      DEBUG    got {'id': 'ralnsy', 'ret': 'ok', 'resp': "<ctl ret='ok' tp='sa' msid='11'><m mid='0' p='1'/><m mid='1' p='1'/><m mid='2' p='1'/><m mid='3' p='1'/><m mid='4' p='1'/><m mid='5' p='1'/><m mid='6' p='1'/><m mid='7' p='1'/><m mid='8' p='1'/></ctl>"}
```
This tells you I have 9 rooms mapped (mid= 0 - 8) or A-I, but you should be able to compare to the map in the app now to know which mid == what room.

### Friendly Names
If you want to rename them to "friendly names" - something the app won't let you do natively:
You need to convert the xml from the GetMapSet to json:
``` xml
<ctl ret='ok' tp='sa' msid='11'><m mid='0' p='1'/><m mid='1' p='1'/><m mid='2' p='1'/><m mid='3' p='1'/><m mid='4' p='1'/><m mid='5' p='1'/><m mid='6' p='1'/><m mid='7' p='1'/><m mid='8' p='1'/></ctl>
```
becomes
``` javascript
{"ctl":{"ret":"ok","tp":"sa","msid":"11","m":[{"mid":"0","p":"1"},{"mid":"1","p":"1"},{"mid":"2","p":"1"},{"mid":"3","p":"1"},{"mid":"4","p":"1"},{"mid":"5","p":"1"},{"mid":"6","p":"1"},{"mid":"7","p":"1"},{"mid":"8","p":"1"}]}}
```
Now you need to add a "n" attribute which contains the friendly name like so:
``` javascript
{"ctl":{"ret":"ok","tp":"sa","msid":"11","m":[{"mid":"0","n":"Entry"},{"mid":"1","n":"Master Bath"},{"mid":"2","n":"Master"},{"mid":"3","n":"Office"},{"mid":"4","n":"Play Room"},{"mid":"5","n":"Craft Room"},{"mid":"6","n":"Kitchen"},{"mid":"7","n":"Sun Room"},{"mid":"8","n":"Garage Entry"}]}}
```
Remove the api response details ("ctl" and "ret"):
``` javascript
{"tp":"sa","msid":"11","m":[{"mid":"0","n":"Entry"},{"mid":"1","n":"Master Bath"},{"mid":"2","n":"Master"},{"mid":"3","n":"Office"},{"mid":"4","n":"Play Room"},{"mid":"5","n":"Craft Room"},{"mid":"6","n":"Kitchen"},{"mid":"7","n":"Sun Room"},{"mid":"8","n":"Garage Entry"}]}
```
Last of all you can use the below command to issue the rename:
```
vacbot.run(VacBotCommand("RenameM", {"tp":"sa","msid":"11","m":[{"mid":"0","n":"Entry"},{"mid":"1","n":"Master Bath"},{"mid":"2","n":"Master"},{"mid":"3","n":"Office"},{"mid":"4","n":"Play Room"},{"mid":"5","n":"Craft Room"},{"mid":"6","n":"Kitchen"},{"mid":"7","n":"Sun Room"},{"mid":"8","n":"Garage Entry"}]}))
```
You should then see the friendly names in the app when selecting an area to clean. 
**Note:** You cannot use the friendly names when starting a clean, you must use the mid.

kquinsland, 2019-08-07:
@wpietri ,  is there anything that's preventing this from being merged in?  I'd like to use the library inside Home Assistant without mucking about with forks in my HA python setup.

addd45, 2019-12-08:
@kquinsland do you have a workaround going? I'm curious how youre doing it

poldim, 2020-07-10:
If anyone has a Deebot 900 and wants to have some control (not all) through Home Assistant, check out these options:

I'm using the one by [Ligio](https://github.com/Ligio/hacc-ozmo
), and have tried another by [Soltroy](https://gist.github.com/Soltroy/b3ae9a48f4030e3d00fda22af52723d8
).  Many things work but a few do not. YMMV