wpietri/sucks #39: Not able to clean

MonsterCode8000, 2018-06-04:
After I have lured myself in with "sucks login" and then start the cleaning program (eg. clean 1) error messages appear.

```
ERROR    call to GetDeviceList failed with {'todo': 'result', 'result': 'fail', 'errno': 1004, 'error': 'auth error'}
Traceback (most recent call last):
  File "/usr/local/bin/sucks", line 11, in <module>
    load_entry_point('sucks==0.8.4', 'console_scripts', 'sucks')()
  File "/home/nick/.local/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/nick/.local/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/nick/.local/lib/python3.6/site-packages/click/core.py", line 1093, in invoke
    return _process_result(rv)
  File "/home/nick/.local/lib/python3.6/site-packages/click/core.py", line 1031, in _process_result
    **ctx.params)
  File "/home/nick/.local/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/nick/.local/lib/python3.6/site-packages/sucks/cli.py", line 211, in run
    vacuum = api.devices()[0]
  File "/home/nick/.local/lib/python3.6/site-packages/sucks/__init__.py", line 112, in devices
    'resource': self.resource
  File "/home/nick/.local/lib/python3.6/site-packages/sucks/__init__.py", line 93, in __call_user_api
    "failure {} ({}) for call {} and parameters {}".format(json['error'], json['errno'], function, params))
RuntimeError: failure auth error (1004) for call GetDeviceList and parameters {'todo': 'GetDeviceList', 'userid': '20180511213956_c...............', 'auth': {'with': 'users', 'userid': '20180511213956_c......', 'realm': 'ecouser.net', 'token': 'fQVh----', 'resource': 'b0d3524a'}}

```

I read that some problems are existing with the "GetDeviceList" but how else should I start cleaning??

arronhunt, 2018-07-10:
I'm getting this issue as well

wpietri, 2018-07-17:
If someone is still getting this, could they please do the same command with "--debug" mixed in and attach the whole output? I just cleared my config file and did everything fresh, and it's working fine for me.

vietquocnguyen, 2018-07-17:
Here you go @wpietri. FYI I have a **N79S**
```
Viets-MBP:~ vietnguyen$ sucks login
Ecovacs app email: viet@vqndev.com
Ecovacs app password:
your two-letter country code [us]:
your two-letter continent code [na]:
overwrite existing config? [y/N]: y
Config saved.
Viets-MBP:~ vietnguyen$ sucks --debug clean 10
DEBUG    will run [<sucks.cli.CliAction object at 0x10bb140b8>, <sucks.cli.CliAction object at 0x10bb14128>]
DEBUG    Setting up EcoVacsAPI
DEBUG    calling main api user/login with (('account', '1fzDi5sDjSK03sGjh/3ziTHG3OTFnzJo3EHGM7mRKnNqTX+Qr/A5cdcYZxhe4IkFQ0misvO0lkTHdfvHxhTVFbDoD845RRwUhw926RFlni0Zrsi1ctPa90FC/i9xI0U7+HWEx2oR4wwGC387denTvVeAMro0ta2usAIIkA4+XXg='), ('password', 'Lvf6hzeqm2+RWkZLarunbave8uePr7cKByJ4JSZGgEWHHPAWF8+cUxTQ1tIrrcNERcI4E0uLhpD4FA0Ttnc9LcL5kwHrV6XRfMn9l5Raz/9FTGz3MvF+F0sxSgDuIv468HKgSdqw19Mzwq2RY+Gq0NPaL6Oriq9LhEHtVFPHlDQ='))
DEBUG    Starting new HTTPS connection (1): eco-us-api.ecovacs.com:443
DEBUG    https://eco-us-api.ecovacs.com:443 "GET /v1/private/us/en/2417f45431d41d875774320a0943a84e/i_eco_e/1.3.5/c_googleplay/1/user/login?account=1fzDi5sDjSK03sGjh%2F3ziTHG3OTFnzJo3EHGM7mRKnNqTX%2BQr%2FA5cdcYZxhe4IkFQ0misvO0lkTHdfvHxhTVFbDoD845RRwUhw926RFlni0Zrsi1ctPa90FC%2Fi9xI0U7%2BHWEx2oR4wwGC387denTvVeAMro0ta2usAIIkA4%2BXXg%3D&password=Lvf6hzeqm2%2BRWkZLarunbave8uePr7cKByJ4JSZGgEWHHPAWF8%2BcUxTQ1tIrrcNERcI4E0uLhpD4FA0Ttnc9LcL5kwHrV6XRfMn9l5Raz%2F9FTGz3MvF%2BF0sxSgDuIv468HKgSdqw19Mzwq2RY%2BGq0NPaL6Oriq9LhEHtVFPHlDQ%3D&requestId=cddc269d6ed369b333812e23c44d3b86&authTimespan=1531854310018&authTimeZone=GMT-8&authAppkey=eJUWrzRv34qFSaYk&authSign=c8f940a5101f0856131d4d2c740eaf95 HTTP/1.1" 200 None
DEBUG    got {'code': '0000', 'msg': '操作成功', 'time': 1531854310179, 'data': {'uid': '20180410145022_8fa66d5757f8e93e11647a7a5bb67abc', 'username': 'EA4B5YEH', 'email': 'viet@vqndev.com', 'country': 'us', 'accessToken': '964a3cea28847d689869f85377afc614'}}
DEBUG    calling main api user/getAuthCode with (('uid', '20180410145022_8fa66d5757f8e93e11647a7a5bb67abc'), ('accessToken', '964a3cea28847d689869f85377afc614'))
DEBUG    Starting new HTTPS connection (1): eco-us-api.ecovacs.com:443
DEBUG    https://eco-us-api.ecovacs.com:443 "GET /v1/private/us/en/2417f45431d41d875774320a0943a84e/i_eco_e/1.3.5/c_googleplay/1/user/getAuthCode?uid=20180410145022_8fa66d5757f8e93e11647a7a5bb67abc&accessToken=964a3cea28847d689869f85377afc614&requestId=6b45a98986e730e88dfa31c8a2648caf&authTimespan=1531854310246&authTimeZone=GMT-8&authAppkey=eJUWrzRv34qFSaYk&authSign=47b18653c5d331da8247af437ed8473e HTTP/1.1" 200 None
DEBUG    got {'code': '0000', 'msg': '操作成功', 'time': 1531854310398, 'data': {'authCode': 'us_c446350fa3316d49b11513dbdb392978', 'ecovacsUid': '20180411025040_ec4f6d30a653b778382537761a171575'}}
DEBUG    calling user api loginByItToken with {'country': 'US', 'resource': '2417f454', 'realm': 'ecouser.net', 'userId': '20180410145022_8fa66d5757f8e93e11647a7a5bb67abc', 'token': 'us_c446350fa3316d49b11513dbdb392978'}
DEBUG    Starting new HTTPS connection (1): users-na.ecouser.net:8000
DEBUG    https://users-na.ecouser.net:8000 "POST /user.do HTTP/1.1" 200 124
DEBUG    got {'todo': 'result', 'result': 'ok', 'userId': 'cdksopb38a816c89', 'resource': '2417f454', 'token': 'h26vAMVhQ5m9kSUtWfnR0P8xxUHTd51X'}
DEBUG    EcoVacsAPI connection complete
DEBUG    calling user api GetDeviceList with {'userid': '20180410145022_8fa66d5757f8e93e11647a7a5bb67abc', 'auth': {'with': 'users', 'userid': '20180410145022_8fa66d5757f8e93e11647a7a5bb67abc', 'realm': 'ecouser.net', 'token': 'h26vAMVhQ5m9kSUtWfnR0P8xxUHTd51X', 'resource': '2417f454'}}
DEBUG    Starting new HTTPS connection (1): users-na.ecouser.net:8000
DEBUG    https://users-na.ecouser.net:8000 "POST /user.do HTTP/1.1" 200 67
DEBUG    got {'todo': 'result', 'result': 'fail', 'errno': 1004, 'error': 'auth error'}
ERROR    call to GetDeviceList failed with {'todo': 'result', 'result': 'fail', 'errno': 1004, 'error': 'auth error'}
Traceback (most recent call last):
  File "/Library/Frameworks/Python.framework/Versions/3.6/bin/sucks", line 11, in <module>
    sys.exit(cli())
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/click/core.py", line 1093, in invoke
    return _process_result(rv)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/click/core.py", line 1031, in _process_result
    **ctx.params)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/sucks/cli.py", line 211, in run
    vacuum = api.devices()[0]
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/sucks/__init__.py", line 164, in devices
    'resource': self.resource
  File "/Library/Frameworks/Python.framework/Versions/3.6/lib/python3.6/site-packages/sucks/__init__.py", line 145, in __call_user_api
    "failure {} ({}) for call {} and parameters {}".format(json['error'], json['errno'], function, params))
RuntimeError: failure auth error (1004) for call GetDeviceList and parameters {'todo': 'GetDeviceList', 'userid': '20180410145022_8fa66d5757f8e93e11647a7a5bb67abc', 'auth': {'with': 'users', 'userid': '20180410145022_8fa66d5757f8e93e11647a7a5bb67abc', 'realm': 'ecouser.net', 'token': 'h26vAMVhQ5m9kSUtWfnR0P8xxUHTd51X', 'resource': '2417f454'}}
Viets-MBP:~ vietnguyen$
```

Nowaker, 2018-07-18:
@vietquocnguyen `--debug` options seems to expose your credentials in a likely base64ed form. I suggest that you change your credentials ASAP.

wpietri, 2018-07-22:
I think this is the same as #33, so let's shift discussion there.

closed 2018-07-22 by wpietri

vietquocnguyen, 2018-08-28:
Thank you @Nowaker