wpietri/sucks #42: Installation issue

kbaxter, 2018-07-15:
Hello,

I've successfully installed sucks on Raspbian with Python 3.5.3. When I try to invoke login ("sucks login"), I get "-bash: sucks: command not found". 

I've confirmed that it is installed for Python3 using "pip3 freeze".

I am new to Python, so I may be doing something entirely wrong. Any ideas?

Thanks,

Kyle

wpietri, 2018-07-17:
Hi, Kyle. I don't know anything about Rasbian or how you installed sucks, so it's hard for me to say. But you could try running `pip3 show --files sucks` to see where it put the files. Hopefully one of them is just called `sucks`.

On my system, that is a relative path, and it's relative not to the current directory, but the `Location:` line in the pip3 output.

You could also try doing something like 'find / --name sucks` to look everywhere for the file.

closed 2018-07-17 by wpietri

kbaxter, 2018-07-21:
Thanks. `pip3 show --files sucks` returns a location of "/home/pi/.local/lib/python3.5/site-packages" and files:
"../../../bin/sucks
  sucks-0.8.4.dist-info/DESCRIPTION.rst
  sucks-0.8.4.dist-info/INSTALLER
  sucks-0.8.4.dist-info/METADATA
  sucks-0.8.4.dist-info/RECORD
  sucks-0.8.4.dist-info/WHEEL
  sucks-0.8.4.dist-info/entry_points.txt
  sucks-0.8.4.dist-info/metadata.json
  sucks-0.8.4.dist-info/top_level.txt
  sucks/__init__.py
  sucks/__pycache__/__init__.cpython-35.pyc
  sucks/__pycache__/cli.cpython-35.pyc
  sucks/cli.py"

It seems to be installed correctly, so I'm not really sure what's going on.

wpietri, 2018-07-21:
So does that mean that the sucks command itself in /home/pi/.local/bin? If so, you should be able to run it from there. And if you want to be able to run a command without typing the full path, you can change your PATH shell variable to accommodate. More here: https://kb.iu.edu/d/acar

mrjcohen, 2020-01-04:
I'm running into the same issue. Did you ever figure this out? Tried running 'sucks login' directly from ' /home/pi/.local/bin' but it says "bash: sucks: command not found"

Running 'pip show --files sucks' says that its there

mrjcohen, 2020-01-04:
I was adding the PATH incorrectly so now sucks command is recognized but now I am having another problem:

`hoobs@hoobs:~/.local/bin $ sucks login
Traceback (most recent call last):
  File "/home/hoobs/.local/bin/sucks", line 6, in <module>
    from sucks.cli import cli
  File "/home/hoobs/.local/lib/python2.7/site-packages/sucks/__init__.py", line 360
    def is_charging(self) -> bool:
                          ^
SyntaxError: invalid syntax`

Im a noob so I am completely clueless