wpietri/sucks #45: sleekxmpp.basexmpp ERROR    day is out of range for month

btittelbach, 2018-08-06:
* Login works
* Server used: eu
* any further command throws the following error

```
sleekxmpp.basexmpp WARNING  fulljid property deprecated. Use boundjid.resource                                          
sleekxmpp.basexmpp ERROR    day is out of range for month
Traceback (most recent call last):                                                                                    
  File "/home/bernhard/.local/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1492, in _process
    if not self.__read_xml():                                                                                      
  File "/home/bernhard/.local/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1564, in __read_xml
    self.__spawn_event(xml)                                                                                                                     
  File "/home/bernhard/.local/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1632, in __spawn_event
    handler.prerun(stanza_copy)                                                                                    
  File "/home/bernhard/.local/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
    self.run(payload, True)                                                                                
  File "/home/bernhard/.local/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)                                                                                        
  File "/home/bernhard/.local/lib/python3.6/site-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
    if self.xmpp.start_tls():                                            
  File "/home/bernhard/.local/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 889, in start_tls     
    cert.verify(self._expected_server_name, self._der_cert)     
  File "/home/bernhard/.local/lib/python3.6/site-packages/sleekxmpp/xmlstream/cert.py", line 133, in verify           
    not_before, not_after = extract_dates(raw_cert)
  File "/home/bernhard/.local/lib/python3.6/site-packages/sleekxmpp/xmlstream/cert.py", line 111, in extract_dates 
    not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
  File "/usr/lib/python3.6/_strptime.py", line 565, in _strptime_datetime                                                                       
    tt, fraction = _strptime(data_string, format)
  File "/usr/lib/python3.6/_strptime.py", line 528, in _strptime                                                   
    datetime_date(year, 1, 1).toordinal() + 1              
ValueError: day is out of range for month
```

TonyFeestneus, 2018-08-30:
Look at my post here: https://github.com/home-assistant/home-assistant/issues/16186

Can be fixed, but perhaps you will run in to other issues later, like I did.

OverloadUT, 2018-08-30:
More to the point, it seems that this issue is upstream in sleekXMPP: https://github.com/fritzy/SleekXMPP/issues/478

Considering how long that issue has been open, we may need to figure out a workaround here.

SwapFaceL, 2018-09-28:
Any updates on this one? I got the same error and wonder if there is a fix for it.

dgomes, 2018-10-31:
Also finding the same error

lollo78, 2018-11-02:
Same issue of the OP in Home Assistant 0.81.2

h4nc, 2018-11-05:
Same issue here running hassio 0.81.6

kmlucy, 2018-11-11:
If I fix the date out of range error via https://github.com/fritzy/SleekXMPP/issues/478#issuecomment-367328327, I get this error instead:

```
2018-11-10 20:56:07 INFO (read_thread) [sleekxmpp.xmlstream.xmlstream] Negotiating TLS
2018-11-10 20:56:07 INFO (read_thread) [sleekxmpp.xmlstream.xmlstream] Using SSL version: TLSv1
2018-11-10 20:56:07 ERROR (read_thread) [sleekxmpp.xmlstream.xmlstream] Connection error.
2018-11-10 20:56:07 ERROR (read_thread) [sleekxmpp.basexmpp] <TagSet object at 0x7f44641cdba8 tags 0:32:16> not in asn1Spec: <OctetString schema object at 0x7f43d05314a8 tagSet <TagSet object at 0x7f446417c908 tags 0:0:4> encoding iso-8859-1>
Traceback (most recent call last):
  File "/usr/local/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1492, in _process
    if not self.__read_xml():
  File "/usr/local/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1564, in __read_xml
    self.__spawn_event(xml)
  File "/usr/local/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1632, in __spawn_event
    handler.prerun(stanza_copy)
  File "/usr/local/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
    self.run(payload, True)
  File "/usr/local/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/usr/local/lib/python3.6/site-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
    if self.xmpp.start_tls():
  File "/usr/local/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 889, in start_tls
    cert.verify(self._expected_server_name, self._der_cert)
  File "/usr/local/lib/python3.6/site-packages/sleekxmpp/xmlstream/cert.py", line 140, in verify
    cert_names = extract_names(raw_cert)
  File "/usr/local/lib/python3.6/site-packages/sleekxmpp/xmlstream/cert.py", line 73, in extract_names
    asn1Spec=OctetString())[0]
  File "/usr/local/lib/python3.6/site-packages/pyasn1/codec/ber/decoder.py", line 1318, in __call__
    '%s not in asn1Spec: %r' % (tagSet, asn1Spec)
pyasn1.error.PyAsn1Error: <TagSet object at 0x7f44641cdba8 tags 0:32:16> not in asn1Spec: <OctetString schema object at 0x7f43d05314a8 tagSet <TagSet object at 0x7f446417c908 tags 0:0:4> encoding iso-8859-1>
2018-11-10 20:56:07 INFO (read_thread) [sleekxmpp.xmlstream.xmlstream] Waiting for </stream:stream> from server
```

alagarath, 2018-11-24:
@kmlucy 
Just bought one of these off Amazon on sale and I'm getting the same error after I tried the fix from that [thread](https://github.com/fritzy/SleekXMPP/issues/478#issuecomment-367328327). For now I've just removed the component from my configuration.yaml because it's a bit chatty in the Home Assistant  logs and it delays reboots. Looking forward to seeing how this develops.

[Line 111](https://github.com/fritzy/SleekXMPP/blob/7075a3b7fa857fc8f9db60da9d01fa1a11806659/sleekxmpp/xmlstream/cert.py#L111)
`not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')`

Replaced with...
`    try:
        not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
    except ValueError:
        not_before = datetime.strptime(not_before, '%y%m%d%H%M%SZ')`

[Line 115](https://github.com/fritzy/SleekXMPP/blob/7075a3b7fa857fc8f9db60da9d01fa1a11806659/sleekxmpp/xmlstream/cert.py#L115)
`not_after = datetime.strptime(not_after, '%Y%m%d%H%M%SZ')`

Replaced with...
`not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
    try:
        not_after = datetime.strptime(not_after, '%Y%m%d%H%M%SZ')
    except ValueError:
        not_after = datetime.strptime(not_after, '%y%m%d%H%M%SZ')`

kmlucy, 2018-11-24:
@alagarath I actually ended up just reverting to the previous dependencies, which fixed the problem for the moment. It's not a long term solution, but it fixes the issue for the moment: https://github.com/home-assistant/home-assistant/issues/18054#issuecomment-438657231

h4nc, 2018-11-24:
Would be nice if the working (latest) dependencies will be reported in this topic by the first one who see's it working.

jpreiditsch, 2018-11-29:
Seeing this on 0.82.1

marlonalkan, 2018-12-01:
> @kmlucy
> Just bought one of these off Amazon on sale and I'm getting the same error after I tried the fix from that [thread](https://github.com/fritzy/SleekXMPP/issues/478#issuecomment-367328327). For now I've just removed the component from my configuration.yaml because it's a bit chatty in the Home Assistant logs and it delays reboots. Looking forward to seeing how this develops.
> 
> [Line 111](https://github.com/fritzy/SleekXMPP/blob/7075a3b7fa857fc8f9db60da9d01fa1a11806659/sleekxmpp/xmlstream/cert.py#L111)
> `not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')`
> 
> Replaced with...
> 
> ```
>    try:
>         not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
>     except ValueError:
>         not_before = datetime.strptime(not_before, '%y%m%d%H%M%SZ')
> ```
> [Line 115](https://github.com/fritzy/SleekXMPP/blob/7075a3b7fa857fc8f9db60da9d01fa1a11806659/sleekxmpp/xmlstream/cert.py#L115)
> `not_after = datetime.strptime(not_after, '%Y%m%d%H%M%SZ')`
> 
> Replaced with...
> 
> ```
>     try:
>         not_after = datetime.strptime(not_after, '%Y%m%d%H%M%SZ')
>     except ValueError:
>         not_after = datetime.strptime(not_after, '%y%m%d%H%M%SZ')
> ```

Hey, did you managed to get it working ?

alagarath, 2018-12-01:
@marlonalkan
Yes, but only by rolling back to the previous dependencies mentioned by kmlucy.

I ran the following in my docker container terminal from the [link](https://github.com/home-assistant/home-assistant/issues/18054#issuecomment-438657231) kmlucy posted
`RUN pip uninstall -y pyasn1 pyasn1-modules sleekxmpp slixmpp && pip install pyasn1==0.3.7 pyasn1 modules==0.1.5 sleekxmpp==1.3.2 dnspython3==1.15.0`

inputd, 2018-12-06:
I've read the workarounds you guys posted, but it seems to me its for non-hass.io versions of HA. Has anyone found success adopting the workaround in Hass.io?

rschoolm, 2018-12-12:
I'm in the same boat.  How do we work around this for HASSIO?

alagarath, 2018-12-12:
I wish I had some experience with hassio. I would imagine it's using the same sort of dependencies somewhere in the virtual directory for hassio. Unfortunately I have no experience atm

rschoolm, 2018-12-12:
Thanks alagarth.  Appreciate the quick response.  We will have to see what comes about with SleekXMPP.

rschoolm, 2018-12-12:
Is this a BUG with the sleekxmpp or is the ecovac addon using an outdated or depreciated method within sleekxmpp?

wpietri, 2018-12-12:
For what it's worth, I do a clean download-and-build with this code every day to run my vacuum. I don't get this error ever. So I believe this is something specific to HomeAssistant users. I'm happy to take a patch that fixes the problem, though.

wpietri, 2018-12-12:
Looking at the logs from the last nightly build, I see this: `Successfully installed atomicwrites-1.2.1 attrs-18.2.0 certifi-2018.11.29 chardet-3.0.4 click-7.0 coverage-4.5.2 idna-2.7 more-itertools-4.3.0 pathlib2-2.3.3 pluggy-0.8.0 pprintpp-0.4.0 py-1.7.0 pycountry-18.12.8 pycountry-convert-0.7.2 pycryptodome-3.7.2 pytest-4.0.1 pytest-cov-2.6.0 pytest-mock-1.10.0 repoze.lru-0.7 requests-2.20.1 scandir-1.9.0 six-1.11.0 sleekxmpp-1.3.3 stringcase-1.2.0 sucks-0.9.3 urllib3-1.24.1`

kmlucy, 2018-12-12:
Home Assistant has moved from sleekxmpp to slixmpp==1.4.1. I believe this is where the problems comes from. I, and others, have been able to get it working again by reverting to sleekxmpp temporarily.

OverloadUT, 2018-12-12:
@wpietri: The issue is related to the way timestamps are coming from or being sent to the XMPP server. I find it entirely likely that this could be an issue experienced differently on different combinations of OS and hardware. On my dev machine, sucks works fine via Home Assistant, but on my production Raspberry Pi, it does not. :(

If you look at the comments on the issue in SleekXMPP you can some other users outside of Home Assistant that have been experiencing this issue: https://github.com/fritzy/SleekXMPP/issues/478

The second comment has a person experiencing it on only one out of a pool of servers, so there's something weird going on.

There are a bunch more people reporting having the issue here in a pull request that was rejected: https://github.com/fritzy/SleekXMPP/issues/461

I don't know enough about XMPP to make an authoritative call here, which is the only reason I haven't pinned sucks to XMPP 1.3.2 (before the issue was introduced.)

rschoolm, 2018-12-12:
Wow!  What a community!

I was digging around @fritzy/SleekXMPP some this AM and I saw the same thing.  I really wish I could contribute here, but I'm not a developer.  I do know scripting languages though and currently learning Python.

I do want to say thank you for the time and effort to put this addon together.

rschoolm

OverloadUT, 2018-12-12:
> Home Assistant has moved from sleekxmpp to slixmpp==1.4.1

@kmlucy Can you link me to where discussion on this suspicion is taking place? I'd like to dig in on why slixmpp might be affecting what SleekXMPP is doing.

leofuscaldi, 2019-01-22:
I'm using HassIO and I don't know how to go back to 1.3.2 or 1.3.1... can someone point me on how to do it?

trhr, 2019-01-27:
i did some digging and noticed that the xmpp is a pretty minor part of this library...

can someone with better python chops just swap the xmpp library to one that works?

wpietri, 2019-01-27:
All of the communication with the vacuum itself is over XMPP, so in some sense it's the heart of it. When I wrote this I looked at a variety of options, and this seemed like the best of a shaky lot. But if somebody would like to take a swing at trying a different library, I'd be very interested to see how it turns out.

TogueSoftware, 2019-02-14:
Same problem with XMPP 1.3.3

Edit the file  _strptime.py located in ..\Python\Lib\

And add the folowing code before the line 527 (starting with # If we know the week )

    if month == 2 and day > 28 :
        day = 28

h4nc, 2019-02-15:
@tongueSoftware does this solve this issue, would be great because this also destroyed the ecovacs components for some months.

slmi, 2019-02-17:
for sleekxmpp-1.3.3

```
--- sleekxmpp/xmlstream/cert.py
+++ sleekxmpp/xmlstream/cert.py	2019-02-17 16:09:21.738607425 +0100
@@ -108,11 +108,20 @@
 
     not_before = validity.getComponentByName('notBefore')
     not_before = str(not_before.getComponent())
-    not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
+
+    if not_before.__len__() == 13:
+        not_before = datetime.strptime(not_before, '%y%m%d%H%M%SZ')
+    else:
+        not_before = datetime.strptime(not_before, '%Y%m%d%H%M%SZ')
 
     not_after = validity.getComponentByName('notAfter')
     not_after = str(not_after.getComponent())
-    not_after = datetime.strptime(not_after, '%Y%m%d%H%M%SZ')
+
+    if not_after.__len__() == 13:
+        not_after = datetime.strptime(not_after, '%y%m%d%H%M%SZ')
+    else:
+        not_after = datetime.strptime(not_after, '%Y%m%d%H%M%SZ')
+
 
     return not_before, not_after

```

Frippuz, 2019-03-06:
@slmi I've tried the above code and the date error seems to be fixed. But...
I get this instead:
`Mar 06 13:02:17 hassbian hass[8366]: 2019-03-06 13:02:17 ERROR (read_thread) [sleekxmpp.xmlstream.xmlstream] Connection error.
Mar 06 13:02:17 hassbian hass[8366]: 2019-03-06 13:02:17 ERROR (read_thread) [sleekxmpp.basexmpp] <TagSet object at 0x6a696850 tags 0:32:16> not in asn1Spec: <OctetString schema object at 0x646fe350 tagSet <TagSet object at 0x69e5eb90 tags 0:0:4> encoding iso-8859-1>
Mar 06 13:02:17 hassbian hass[8366]: Traceback (most recent call last):
Mar 06 13:02:17 hassbian hass[8366]: File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1492, in _process
Mar 06 13:02:17 hassbian hass[8366]: if not self.__read_xml():
Mar 06 13:02:17 hassbian hass[8366]: File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1564, in __read_xml
Mar 06 13:02:17 hassbian hass[8366]: self.__spawn_event(xml)
Mar 06 13:02:17 hassbian hass[8366]: File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1632, in __spawn_event
Mar 06 13:02:17 hassbian hass[8366]: handler.prerun(stanza_copy)
Mar 06 13:02:17 hassbian hass[8366]: File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 64, in prerun
Mar 06 13:02:17 hassbian hass[8366]: self.run(payload, True)
Mar 06 13:02:17 hassbian hass[8366]: File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
Mar 06 13:02:17 hassbian hass[8366]: self._pointer(payload)
Mar 06 13:02:17 hassbian hass[8366]: File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/features/feature_starttls/starttls.py", line 64, in _handle_starttls_proceed
Mar 06 13:02:17 hassbian hass[8366]: if self.xmpp.start_tls():
Mar 06 13:02:17 hassbian hass[8366]: File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 889, in start_tls
Mar 06 13:02:17 hassbian hass[8366]: cert.verify(self._expected_server_name, self._der_cert)
Mar 06 13:02:17 hassbian hass[8366]: File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/cert.py", line 144, in verify
Mar 06 13:02:17 hassbian hass[8366]: cert_names = extract_names(raw_cert)
Mar 06 13:02:17 hassbian hass[8366]: File "/srv/homeassistant/lib/python3.6/site-packages/sleekxmpp/xmlstream/cert.py", line 73, in extract_names
Mar 06 13:02:17 hassbian hass[8366]: asn1Spec=OctetString())[0]
Mar 06 13:02:17 hassbian hass[8366]: File "/srv/homeassistant/lib/python3.6/site-packages/pyasn1/codec/ber/decoder.py", line 1318, in __call__
Mar 06 13:02:17 hassbian hass[8366]: '%s not in asn1Spec: %r' % (tagSet, asn1Spec)
Mar 06 13:02:17 hassbian hass[8366]: pyasn1.error.PyAsn1Error: <TagSet object at 0x6a696850 tags 0:32:16> not in asn1Spec: <OctetString schema object at 0x646fe350 tagSet <TagSet object at 0x69e5eb90 tags 0:0:4> encoding iso-8859-1>`

slmi, 2019-03-06:
[https://github.com/fritzy/SleekXMPP/pull/482/files](url)

```
--- sleekxmpp/xmlstream/cert.py
+++ sleekxmpp/xmlstream/cert.py	2019-03-06 14:32:21.715036120 +0100
@@ -69,8 +69,7 @@
         if oid != SUBJECT_ALT_NAME:
             continue
 
-        value = decoder.extension.getComponentByName('extnValue'),
-                               asn1Spec=OctetString())[0]
+        value = extension.getComponentByName('extnValue')
         sa_names = decoder.decode(value, asn1Spec=SubjectAltName())[0]
         for name in sa_names:
             name_type = name.getName()

```

Frippuz, 2019-03-06:
@slmi I'm getting closer :-)
New error thou. But I wonder if this has to do with a sleekxmpp problem.

Mar 06 14:59:00 hassbian hass[9475]: 2019-03-06 14:59:00 ERROR (read_thread) [sleekxmpp.xmlstream.xmlstream] Socket Error #9: Bad file descriptor
Mar 06 14:59:06 hassbian hass[9475]: 2019-03-06 14:59:06 ERROR (read_thread) [sleekxmpp.xmlstream.xmlstream] Certificate has expired.
Mar 06 14:59:06 hassbian hass[9475]: 2019-03-06 14:59:06 WARNING (read_thread) [sleekxmpp.xmlstream.xmlstream] Failed to send b"<stream:stream to='ecouser.net' xmlns:stream='http://etherx.jabber.org/streams' xmlns='jabber:client' xml:lang='en' version='1.0'>"

I'm also getting this: 
Mar 06 14:58:07 hassbian hass[9475]: 2019-03-06 14:58:07 WARNING (SyncWorker_4) [sleekxmpp.basexmpp] fulljid property deprecated. Use boundjid.resource

coolguymatt, 2019-04-05:
@Frippuz 

Did you figure this out? I have the same issue.

meyerdenney, 2019-04-07:
@Frippuz - how are you pushing these changes to hassio? I'm getting the same error but only on my Raspberry Pi 3 that is running hassio. I would like to try and make a fix myself but I'm not sure how would I push the change to hassio.

Neustradamus, 2019-04-08:
The main code works no?
- https://github.com/fritzy/SleekXMPP/commits/develop

inputd, 2019-04-08:
> @Frippuz - how are you pushing these changes to hassio? I'm getting the same error but only on my Raspberry Pi 3 that is running hassio. I would like to try and make a fix myself but I'm not sure how would I push the change to hassio.

AFAIK, there is no work around for hassio yet =/

Neustradamus, 2019-04-08:
Do you need a 1.3.4 of SleekXMPP?

Frippuz, 2019-04-08:
It seems that the fixes above helps. But it also seems like I have issues with two libs at the same time. I'm trying to get sucks to work. But it looks like the error wasn't with sucks in the end anyway. My type of 'bot wasn't supported by sucks at all - something that wasn't clear when In started testing it all. There are solutions on the way for that part.

I got stuck with the error above. But I think it is more related to sucks and not SleekXMPP itself. To resolve my original error, I downgraded to 1.3.2 and removed the certificate check. I am also running an experimental version of sucks and it works - unsafe, but it works at least. 

Check this out: https://github.com/bmartin5692/sucks/tree/D901

Neustradamus, 2019-04-08:
@Frippuz: Thanks for your reply!
If you test with https://github.com/fritzy/SleekXMPP/commits/develop instead of 1.3.2?

OverloadUT, 2019-04-08:
I spent some time last night finally sitting down to truly debug this. I ran into several issues:
1. The HEAD of develop did not work out of the box at all, because it has invalid references to some deprecated properties of the SSL library. Hassio runs on Python 3.7 where this just blows up. I saw that a PR has been open in SleekXMPP for this for ages
2. Once I fixed it to not use the deprecated properties, I ran in to certificate expiration errors. This does mean that the "day is out of range" issue is fixed in develop, but the certificate expiration is a blocker.
3. I tried changing some parameters when SleekXMPP is created (such as not to use TLS at all) but these don't work. We need to have a way to use the secure connection, but to not error out if the cert is invalid, which is currently not a feature that is supported in SleekXMPP. This issue is described in #62 
4. Once I commented out the code in SleekXMPP that does expiration validation, everything worked!

I am wondering if it makes sense for us to try switching to Slixmpp, which is a form of SleekXMPP (so hopefully compatible without too many changes) that has had more recent development. It's either that or I will publish a version of SleekXMPP just for sucks that has patches for these issues.

wpietri, 2019-04-10:
I have no attachment to any particular XMPP library, and this one has been a thorn in our sides for a long time now, so I'm happy to switch to anything that works better.

Neustradamus, 2019-04-10:
@OverloadUT: Please create PR for your changes...

I try to relaunch the dev of the lib, I wait changes from @fritzy and @bear since several months, for a best SleekXMPP...

OverloadUT, 2019-04-10:
@Neustradamus The problem is that I don't have clean changes. I had to just comment out busted stuff and circumvent the cert expiration checks. It would take some time to fix it in such a manner as to be PR-worthy.