wpietri/sucks #46: Warning: "fulljid property deprecated. Use boundjid.resource"

joe248, 2018-08-30:
Using the Ecovacs component in Home Assistant 0.77 and seeing this in the logs constantly:

`WARNING (SyncWorker_12) [sleekxmpp.basexmpp] fulljid property deprecated.       Use boundjid.resource`

joe248, 2018-08-30:
I also see this but only at startup, so it's not as annoying:

`WARNING (SyncWorker_12) [sleekxmpp.basexmpp] fulljid property deprecated.       Use boundjid.resource`

I'm not sure how to fix this one as I don't see "fulljid" anywhere in the code.

wpietri, 2018-08-30:
Thanks for fixing this! I appreciate it.

closed 2018-08-30 by wpietri

joe248, 2018-08-30:
No problem.  Thanks for this library!

rchamp26, 2019-07-17:
i am running the latest home assistant version 0.95.4 and the ecovacs vacuum now works again, but i continue to get this same warning

fulljid property deprecated. Use boundjid.resource

jumate, 2019-12-07:
Same problem, with home assistant version 0.102.3

bggardner, 2020-05-12:
This is caused by the `sleekxmppfs.basexmpp` module, for which there is no GitHub repo (that I could find).  If you comment out line 610 of `sleekxmppfs/basexmpp.py`, the warning will be suppressed.  The location of the file will vary based on your installation.

@wpietri - I believe the warning should say "resource property deprecated..."  Correspondingly, the use of `self.resource` in this repo should use `self.boundjid.resource` instead.  I may do a pull request.

PR #47 (2018-08-30): Fix warning due to using deprecated property
Fixes #46