wpietri/sucks #48: Does nothing, also no errors thrown

TonyFeestneus, 2018-08-30:
So, after having all the errors that others have descibed in the issues, I managed to get this error free by using pyasn1=0.1.9 and pyasn1-modules=0.1.5, also sleekxmpp=1.3.1.

I have changed the file `__init__.py` to use the shorter userid that you can see when you use command sudo sucks --debug clean 10. The shorter userid gets returned from the servers. If you hardcode that in to the init file, I get to see my device and device name.

After that it stops. (it does give a warning about fulljid being deprecated, but that is it).

How do I proceed from here?

wpietri, 2018-09-04:
I'm sorry it's not working for you.

From that description I have no idea why, so if you're eager to fix the problem I'd suggest snooping on the connnection between a) your vacuum and the official app, and b) your vacuum and this tool. More details on how to do that here:  https://github.com/wpietri/sucks/blob/master/developing.md

TonyFeestneus, 2018-09-05:
I get it. Can you point out where in the code it is hanging when I get to this point, see code:

`sucks      DEBUG    EcoVacsAPI connection complete
sucks      DEBUG    calling user api GetDeviceList with {'auth': {'token': 'xxxxxx', 'userid': 'xxxxxxx', 'realm': 'ecouser.net', 'with': 'users', 'resource': '4911f62d'}, 'userid': 'xxxxxx'}
sucks      DEBUG    got {'todo': 'result', 'result': 'ok', 'devices': [{'class': '115', 'resource': 'atom', 'company': 'eco-legacy', 'nick': 'xxxxxx', 'did': 'xxxxxxxxx', 'name': 'xxxxxxxxx'}]}
sleekxmpp.basexmpp WARNING  fulljid property deprecated. Use boundjid.resource`

and it does nothing from there?

wpietri, 2018-09-05:
What's the command you're running?

For me if I do `sucks stop` the next lines I see after that are:

```
sucks      DEBUG    ----------------- starting session ----------------
sucks      DEBUG    event = {}
performing stop command
```
My guess is that the session_start event isn't happening, meaning that something isn't working with the XMPP interaction with Ecovacs. So if you're looking in the code, I'd start here: https://github.com/wpietri/sucks/blob/master/sucks/__init__.py#L497

TonyFeestneus, 2018-09-05:
I guess so too that the session isn't starting. Perhaps it needs the other user id for that?

wpietri, 2018-09-05:
Here's the whole list of packages in use today for me. (I have a script that automatically installs into a new virtualenv every day so I can make sure it works with the latest.)

```
atomicwrites==1.2.1
attrs==18.2.0
certifi==2018.8.24
chardet==3.0.4
click==6.7
coverage==4.5.1
idna==2.7
more-itertools==4.3.0
pathlib2==2.3.2
pluggy==0.7.1
pprintpp==0.4.0
py==1.6.0
pycountry==18.5.26
pycountry-convert==0.7.2
pycryptodome==3.6.6
pytest==3.7.4
pytest-cov==2.6.0
pytest-mock==1.10.0
repoze.lru==0.7
requests==2.19.1
scandir==1.9.0
six==1.11.0
sleekxmpp==1.3.3
stringcase==1.2.0
sucks==0.8.4
urllib3==1.23
```

If you're having an issue with the XMPP interactions, I can only suggest the same thing I did before:

- capture the network traffic from a working session between the official app and your vacuum
- capture a non-working session between this code and your vacuum
- figure out what the difference is

Then either submit a patch with updated code and protocol docs or post the difference here.