wpietri/sucks #51: Script never exits

Gernot, 2018-09-22:
I'm running `sucks` on a Raspberry Pi and want to trigger it with an iOS 12 Shortcut over SSH. The robot (M88) drives off and does what it's supposed to do, however the script never exits, which confuses the Shortcut. I don't know if that's expected behaviour or the result of all the error messages that `sucks` throws at me, which I don't know how to get rid of:

**Example:** Result of `sucks charge`:

```
sleekxmpp.basexmpp WARNING  fulljid property deprecated. Use boundjid.resource
sleekxmpp.xmlstream.cert WARNING  Could not find pyasn1 and pyasn1_modules. SSL certificate COULD NOT BE VERIFIED.
sleekxmpp.xmlstream.cert WARNING  Could not find pyasn1 and pyasn1_modules. SSL certificate expiration COULD NOT BE VERIFIED.
performing charge command
sleekxmpp.basexmpp WARNING  resource property deprecated. Use boundjid.resource
sleekxmpp.basexmpp WARNING  resource property deprecated. Use boundjid.resource
sleekxmpp.xmlstream.handler.waiter WARNING  Timed out waiting for IqWait_1ff1a9bd-bd01-4983-87a6-4cd56a7dd1f4-3
Traceback (most recent call last):
  File "/usr/local/bin/sucks", line 11, in <module>
    sys.exit(cli())
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 722, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 697, in main
    rv = self.invoke(ctx)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 1093, in invoke
    return _process_result(rv)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 1031, in _process_result
    **ctx.params)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 535, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/sucks/cli.py", line 219, in run
    vacbot.run(action.vac_command)
  File "/home/pi/.local/lib/python3.5/site-packages/sucks/__init__.py", line 412, in run
    self.send_command(action.to_xml())
  File "/home/pi/.local/lib/python3.5/site-packages/sucks/__init__.py", line 409, in send_command
    self.xmpp.send_command(xml, self._vacuum_address())
  File "/home/pi/.local/lib/python3.5/site-packages/sucks/__init__.py", line 478, in send_command
    c.send()
  File "/home/pi/.local/lib/python3.5/site-packages/sleekxmpp/stanza/iq.py", line 233, in send
    raise IqTimeout(self)
sleekxmpp.exceptions.IqTimeout: <iq id="[ID removed]" from="[ID removed] @ecouser.net/[ID removed]" to="[ID removed]@107.ecorobot.net/atom" type="set"><query xmlns="com:ctl"><ctl td="Charge"><charge type="go" /></ctl></query></iq>
sleekxmpp.basexmpp WARNING  resource property deprecated. Use boundjid.resource
sleekxmpp.basexmpp WARNING  resource property deprecated. Use boundjid.resource
sleekxmpp.basexmpp WARNING  resource property deprecated. Use boundjid.resource
[…and so on…]
```

The `sucks stop` command is the only one that exits. Sometimes. I'm running *0.9.2*.

wpietri, 2018-09-23:
Could you include a debugging log? (You may need to redact certain private values before posting it.)

Also, does this behavior happen only when running via ssh? That is, if you log into the Pi and run it directly, does it work fine?

Gernot, 2018-09-23:
Thanks for whatever you changed in *0.9.3*, it seems to have solved this issue, too. Now it works great with Siri Shortcuts!

Just for reference, if anyone is interested, that's the shortcut. Works great, and I have similar shortcuts for charging and stopping.

[image]

closed 2018-09-23 by Gernot