wpietri/sucks #55: Status request

guillebot, 2018-11-22:
This is more a feature request than an issue. I would like to have some kind of status reporting. 

My use case would be controlling the vac from openhab or cron. In order to do that I would like to know first if the machine is fully charged (or else raise an alarm), then make it clean and then make sure that it's back to its base.

I'm experimenting with direct talking via HTTP/XMPP but is really a mess. Would love to have it packed into sucks.

Thanks a lot!

wpietri, 2018-11-22:
Our current theory is that instead of providing full scripting support via the CLI, we have the library code: https://github.com/wpietri/sucks#library-use

In particular, there are the commands `GetCleanState`, `GetChargeState`, and `GetBatteryState`.

I don't know anything about openhab, but I know that Greg Laabs has done a lot of work on this library to make a HomeAssistant integration work, so I'd suggest you join the mailing list, check out the archives, and post describing what you're looking to do. Others might be able to help.

guillebot, 2018-11-22:
Will do that, thanks for answering. Regards