wpietri/sucks #60: Consider a way that multiple vacbots can be controlled

tucknology, 2018-12-02:
Hi @wpietri. Thanks for an awesome API. 

I recently bought a 2nd vacbot (black friday!) and I noticed I can only control one unit at a time using the CLI. It looks like the following line is hardcoded to always choose the first device from GetDevices:

https://github.com/wpietri/sucks/blob/8e4a5ca560d5f6b94a986a495561d7adbc9e9357/sucks/cli.py#L213

I'm wondering whether you'd consider allowing for some arguments to select a vacuum. Some thoughts:
* I treat them as a swarm, so personally I'm ok with issuing any command to every device in the list. I imagine this is easiest to implement.
* Others might like to be able to select a device from the list, but I'm not sure how the EcoVacs reports devices. Is device[0] always the same? Is there a way to get the friendly name I assign in the iOS app (e.g. "Living Room")? How would the user specify the robot? By number? By friendly name? By selecting from a menu?

Thanks for your consideration.

wpietri, 2018-12-02:
Excellent questions all. I don't have more than one vacuum, so I have no strong opinions here. Since the API isn't documented, I also can't really code anything related to this.

But I'd encourage you to take a swing at it if you like, as it seems like a great case to cover. If you'd like to discuss approaches, I suggest joining the mailing list: https://groups.google.com/forum/#!forum/sucks-users

Then you can get feedback from all contributors.

ohkeepa, 2018-12-16:
I have 2 devices, and modify cli.py like this to get them both going at the same time:

```
    if actions:
        config = read_config()
        api = EcoVacsAPI(config['device_id'], config['email'], config['password_hash'],
                         config['country'], config['continent'])
        vacuum = api.devices()[0]
        vacbot0 = VacBot(api.uid, api.REALM, api.resource, api.user_access_token, vacuum, config['continent'])
        vacbot0.connect_and_wait_until_ready()

        vacuum = api.devices()[1]
        vacbot1 = VacBot(api.uid, api.REALM, api.resource, api.user_access_token, vacuum, config['continent'])
        vacbot1.connect_and_wait_until_ready()

        for action in actions:
            click.echo("performing " + str(action.vac_command))
            vacbot0.run(action.vac_command)
            vacbot1.run(action.vac_command)
            action.wait.wait(vacbot0)

        vacbot0.disconnect(wait=True)
        vacbot1.disconnect(wait=True)
```

tucknology, 2019-09-16:
I finally got around to solving this for my particular use case this weekend and I thought I'd share the result. I made my own script which builds on the excellent API. Here it is, feel free to use or modify:

```python
```

closed 2019-09-16 by tucknology