wpietri/sucks #64: Setting the water level in clean command

Nama, 2019-03-16:
Hi,

I'm looking for a way to set the water level (1, 2 or 3). I couldn't find any info how to do so.

bmartin5692, 2019-03-16:
What model do you have? 

This is a guess, but the following might work:
`vacbot.run(VacBotCommand("SetWaterPermeability", {"v": "1"}))`

Nama, 2019-03-16:
DEEBOT OZMO 610

I couldn't test this yet. `sucks clean 10` works, but I have trouble making `vacbot.run(Clean())` to work. Even tho, `vacbot.run(Charge())` works...

I'll try and will answer again, thank you!

bmartin5692, 2019-03-17:
I think once #65 is addressed for you (see my comment there), this will resolve as well.  You should be able to run the set command either before or after a cleaning cycle starts without issue.

Nama, 2019-03-18:
Yes, it works. 1 is lowest, 3 is highest level.
Care to put that info in https://github.com/bmartin5692/sucks/blob/master/protocol.md#cleaning ?

Thank you, very nice from you!

closed 2019-03-19 by Nama

niveksan, 2019-05-23:
Hi bmartin5692,

I'm looking for a way to specify the SpotArea adding an Option for the runtime. e.g. area 1 --15min
I tried the following with OZMO930: sucks area 1 clean 15 but it seems to result in two commands executed one after the other.