wpietri/sucks #65: charge command works, but not clean

Nama, 2019-03-17:
Hi,
My bot: DEEBOT OZMO 610
I looked in every issue (open and closed) but couldn't find anything like this.
I setup sucks yesterday on my RPi using the official server (not bumper) and was able to send the bot charging (for testing purpose I didn't try cleaning). After, I wanted to try the suggestion in #64, than I realized that `vacbot.run(Charge())` is working, but `vacbot.run(Clean())` and `sucks clean 10` are not.

Sometime I get an timeout error (on the charge-command, too)
```
$ sucks clean 1
sleekxmpp.basexmpp WARNING  fulljid property deprecated. Use boundjid.resource
sleekxmpp.xmlstream.cert WARNING  Could not find pyasn1 and pyasn1_modules. SSL certificate COULD NOT BE VERIFIED.
sleekxmpp.xmlstream.cert WARNING  Could not find pyasn1 and pyasn1_modules. SSL certificate expiration COULD NOT BE VERIFIED.
performing clean command
sleekxmpp.xmlstream.handler.waiter WARNING  Timed out waiting for IqWait_**
Traceback (most recent call last):
  File "/home/kirika/venv/sucks/bin/sucks", line 10, in <module>
    sys.exit(cli())
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/click/core.py", line 1164, in invoke
    return _process_result(rv)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/click/core.py", line 1102, in _process_result
    **ctx.params)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/cli.py", line 219, in run
    vacbot.run(action.vac_command)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 416, in run
    self.send_command(action.to_xml())
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 413, in send_command
    self.xmpp.send_command(xml, self._vacuum_address())
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 482, in send_command
    c.send()
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/stanza/iq.py", line 233, in send
    raise IqTimeout(self)
sleekxmpp.exceptions.IqTimeout: <iq id="**" to="**@130.ecorobot.net/atom" from="cgvlut400807eb1d@ecouser.net/**" type="set"><query xmlns="com:ctl"><ctl td="Clean"><clean type="auto" speed="standard" /></ctl></query></iq>
```

If the error doesn't appear on the clean-command, it just goes on, but the bot didn't receive it and just stays still doing nothing.
```
$ sucks --debug clean 1
sucks.cli  DEBUG    will run [<sucks.cli.CliAction object at 0x75daf730>, <sucks.cli.CliAction object at 0x75dafab0>]
sucks      DEBUG    Setting up EcoVacsAPI
# removed these debug-lines
sleekxmpp.basexmpp WARNING  fulljid property deprecated. Use boundjid.resource
sleekxmpp.xmlstream.cert WARNING  Could not find pyasn1 and pyasn1_modules. SSL certificate COULD NOT BE VERIFIED.
sleekxmpp.xmlstream.cert WARNING  Could not find pyasn1 and pyasn1_modules. SSL certificate expiration COULD NOT BE VERIFIED.
sucks      DEBUG    ----------------- starting session ----------------
sucks      DEBUG    event = {}
performing clean command
sucks      DEBUG    Sending command <iq id="**" to="**@130.ecorobot.net/atom" from="cgvlut400807eb1d@ecouser.net/**" type="set"><query xmlns="com:ctl"><ctl td="Clean"><clean type="auto" speed="standard" /></ctl></query></iq>
waiting for 60.0s
sucks      DEBUG    *** sending ping ***
sucks      DEBUG    *** battery_status = 100%sucks      DEBUG    *** battery_status = 100%
sucks      DEBUG    *** sending ping ***
performing charge command
sucks      DEBUG    Sending command <iq id="**" to="E0001076117610040501@130.ecorobot.net/atom" from="cgvlut400807eb1d@ecouser.net/**" type="set"><query xmlns="com:ctl"><ctl td="Charge"><charge type="go" /></ctl></query></iq>
sucks.cli  DEBUG    waiting on charge_status for value charging
sucks      DEBUG    *** charge_status = returning
sucks      DEBUG    *** clean_status = auto fan_speed = high
sucks      DEBUG    *** charge_status = charging
sucks.cli  DEBUG    wait complete; charge_status is now charging
done
```

bmartin5692, 2019-03-17:
@Nama 

Can you try sucks with my D901 branch? https://github.com/bmartin5692/sucks/tree/D901 

I made some changes that allowed for the Ozmo 930 to work, and I assume the Ozmo 610 would need similar changes to work.

Nama, 2019-03-18:
Works! Very nice!

`sucks login` doesn't set verify_ssl in the config. I had to enter it manually.
And in "Library Usage" the first line needs to be `from sucks import *`, since the rest is `EcoVacsAPI` and not `sucks.EcoVacsAPI`.

Ever plan to put these changes in master (in your repository)?

Thank you very much!

bmartin5692, 2019-03-19:
Glad it worked for you.  I pushed some additional commits to address the items you raised.  I've also merged the D901 branch into my master branch.

Nama, 2019-03-19:
Thanks, that's awesome!
I forgot to mention the `vacbot.disconnect(wait=True)`, since the script doesn't exists otherwise.
I think everyone would benefit, if this were in the "Library Usage" example, too.

closed 2019-03-19 by Nama