wpietri/sucks #66: Getting no status on bmartin5692's fork

Nama, 2019-04-17:
I try to get the status of the bot since 2 days, but all I get is None.
I'm creating the issue here, since @bmartin5692 doesn't allow issues on his repo?

This error seems to be like https://github.com/wpietri/sucks/issues/57, but I didn't even see the response, just the commands I sent.

If I do `sucks charge`, it waits, till the bot finds the dock and charges.
On running Clean(), `vacbot.disconnect(wait=True)` didn't wait till it finishes cleaning (no idea if this is intended and only waits for **starting** to clean).
So I assume cli.py can get the status, since it can wait till the bot starts to charge?

```
import os
import configparser
import itertools
import logging
from sucks import *

config_file = os.path.expanduser('~/.config/sucks.conf')

parser = configparser.ConfigParser()
with open(config_file) as fp:
    parser.read_file(itertools.chain(['[global]'], fp), source=config_file)
    config = parser['global']

api = EcoVacsAPI(config['device_id'], config['email'], config['password_hash'],
                         config['country'], config['continent'])
my_vac = api.devices()[0]
vacbot = VacBot(api.uid, api.REALM, api.resource, api.user_access_token, my_vac, config['continent'], monitor=True)
vacbot.connect_and_wait_until_ready()

logger = logging.getLogger('sucks')
logger.setLevel('DEBUG')
#vacbot.run(VacBotCommand("SetWaterPermeability", {"v": "3"}))
vacbot.request_all_statuses()
#vacbot.send_ping()

# Everything bellow returns None
print(getattr(vacbot, 'vacuum_status'))
vacbot.run(GetCleanState())
vacbot.run(GetChargeState())
vacbot.run(GetBatteryState())
# I've tried these a while back, I assume these are wrong anyways
#print(vacbot.run(VacBotCommand("GetError")))
#print(vacbot.run(VacBotCommand("GetChargeState")))
#print(vacbot.run(VacBotCommand("GetBatteryInfo")))
#print(vacbot.run(VacBotCommand("GetVersion", {"name": "FW"})))
#print(vacbot.run(VacBotCommand("GetLog")))

#vacbot.run(Clean())  # start cleaning
vacbot.disconnect(wait=True)
```

bmartin5692, 2019-04-17:
@Nama - Try this branch in my repo: https://github.com/bmartin5692/sucks/tree/add-fixStatus

I merged the changes jasonarends made in PR #61 / his branch (https://github.com/jasonarends/sucks/tree/fixStatus) 

So this has the changes I made to support Ozmo + the fixes for status that Jason made.

Nama, 2019-04-18:
Works, thank you!
Are errors, if the bot is like, stucked, also in `vacuum_status`?
These return the current status, like "charging", "auto" and "returning".
```
getattr(vacbot, 'vacuum_status')
vacbot.vacuum_status
```
These still return None. That's not intended I assume and the exception bellow might be causing this. Can't get the battery state otherwise, or?
```
vacbot.run(GetCleanState())
vacbot.run(GetChargeState())
vacbot.run(GetBatteryState())
```
Since this version, `vacbot.connect_and_wait_until_ready()` produces this output, but the commands I tested, are still working (including `vacbot.vacuum_status`): Clean(), Charge(), Stop().
```
$ python sucks_client_dev.py 
WARNING:sleekxmpp.basexmpp:fulljid property deprecated. Use boundjid.resource
WARNING:sleekxmpp.xmlstream.cert:Could not find pyasn1 and pyasn1_modules. SSL certificate COULD NOT BE VERIFIED.
WARNING:sleekxmpp.xmlstream.cert:Could not find pyasn1 and pyasn1_modules. SSL certificate expiration COULD NOT BE VERIFIED.
ERROR:sleekxmpp.xmlstream.xmlstream:Error processing stream handler: general
Traceback (most recent call last):
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 895, in _handle_ctl
    s(as_dict)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 432, in _handle_ctl
    getattr(self, method)(ctl)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 462, in _handle_clean_report
    type = event['type']
KeyError: 'type'
ERROR:sleekxmpp.stanza.rootstanza:Error handling {jabber:client}iq stanza
Traceback (most recent call last):
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 895, in _handle_ctl
    s(as_dict)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 432, in _handle_ctl
    getattr(self, method)(ctl)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 462, in _handle_clean_report
    type = event['type']
KeyError: 'type'
ERROR:sleekxmpp.basexmpp:'type'
Traceback (most recent call last):
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 895, in _handle_ctl
    s(as_dict)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 432, in _handle_ctl
    getattr(self, method)(ctl)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 462, in _handle_clean_report
    type = event['type']
KeyError: 'type'
```

bmartin5692, 2019-04-18:
Glad it is working.  Could you run with --debug so we can get more information about the data being returned?  I know for the IOTMQ side (D90X) I had to add some additional checks outside of what jasonarends made to handle those specific items [`__init.py__ Lines 787-796`](https://github.com/bmartin5692/sucks/blob/cac8c43fb6028792219d82b12fd82e3099290800/sucks/__init__.py#L787-L796) and since the Ozmo is a more advanced bot it may have differences there too.

I'm not as familiar with the error handling, but it looks like those are not reported in the status.  I believe the library's stance is the user should handle these by subscribing to the event:
Ex:
```python
def subErrorEvent(incomingError):
    # Handle the error or event
    print("Incoming Error: %s" % (incomingError))
        
vacbot.errorEvents.subscribe(subErrorEvent)
```

You can also get a current error list with "GetError":
````python
vacbot.run(VacBotCommand("GetError"))
````

Nama, 2019-04-18:
Looks like only GetCLeanState() causes the exception. There is no answer from the bot in the debug output...

If I do `print(vacbot.run(GetChargeState()))` I get None, but the bot answers. Are the answers in another way, the library didn't implement?

I'm surprised, that all these bots need different code to work ._.

Thanks for the example!

```
DEBUG:sucks:----------------- starting session ----------------
DEBUG:sucks:event = {}
DEBUG:sucks:*** sending ping ***
DEBUG:sucks:Sending command <iq id="*" to="*@130.ecorobot.net/atom" from="*" type="set"><query xmlns="com:ctl"><ctl td="GetCleanState" id="*" /></query></iq>
DEBUG:sucks:Sending command <iq id="*" to="*@130.ecorobot.net/atom" from="*" type="set"><query xmlns="com:ctl"><ctl td="GetChargeState" id="*" /></query></iq>
ERROR:sleekxmpp.xmlstream.xmlstream:Error processing stream handler: general
Traceback (most recent call last):
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 895, in _handle_ctl
    s(as_dict)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 432, in _handle_ctl
    getattr(self, method)(ctl)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 462, in _handle_clean_report
    type = event['type']
KeyError: 'type'
DEBUG:sucks:Sending command <iq id="2f599daf-2b6e-4d3c-9647-dfbdc94e6aca-6" to="*@130.ecorobot.net/atom" from="cgvlut400807eb1d@ecouser.net/2517be80" type="set"><query xmlns="com:ctl"><ctl td="GetBatteryInfo" id="4408423012" /></query></iq>
ERROR:sleekxmpp.stanza.rootstanza:Error handling {jabber:client}iq stanza
Traceback (most recent call last):
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 895, in _handle_ctl
    s(as_dict)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 432, in _handle_ctl
    getattr(self, method)(ctl)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 462, in _handle_clean_report
    type = event['type']
KeyError: 'type'
ERROR:sleekxmpp.basexmpp:'type'
Traceback (most recent call last):
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 895, in _handle_ctl
    s(as_dict)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 432, in _handle_ctl
    getattr(self, method)(ctl)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 462, in _handle_clean_report
    type = event['type']
KeyError: 'type'
DEBUG:sucks:*** charge_status = charging
DEBUG:sucks:Sending command <iq id="*" to="*@130.ecorobot.net/atom" from="*" type="set"><query xmlns="com:ctl"><ctl td="GetLifeSpan" type="Brush" id="*" /></query></iq>
DEBUG:sucks:*** battery_status = 100%
DEBUG:sucks:Sending command <iq id="*" to="*@130.ecorobot.net/atom" from="*" type="set"><query xmlns="com:ctl"><ctl td="GetLifeSpan" type="SideBrush" id="*" /></query></iq>
DEBUG:sucks:*** life_span main_brush = 1.1833333333333333
DEBUG:sucks:Sending command <iq id="*" to="*@130.ecorobot.net/atom" from="*" type="set"><query xmlns="com:ctl"><ctl td="GetLifeSpan" type="DustCaseHeap" id="*" /></query></iq>
DEBUG:sucks:*** life_span side_brush = 1.1833333333333333
DEBUG:sucks:*** life_span filter = 1.1833333333333333
############################## vacbot.run(GetCleanState())
DEBUG:sucks:Sending command <iq id="*" to="*@130.ecorobot.net/atom" from="*" type="set"><query xmlns="com:ctl"><ctl td="GetCleanState" id="*" /></query></iq>
ERROR:sleekxmpp.xmlstream.xmlstream:Error processing stream handler: general
Traceback (most recent call last):
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
[... 15 lines cut ...]
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 432, in _handle_ctl
    getattr(self, method)(ctl)
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sucks/__init__.py", line 462, in _handle_clean_report
    type = event['type']
KeyError: 'type'
ERROR:sleekxmpp.basexmpp:'type'
Traceback (most recent call last):
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/xmlstream.py", line 1686, in _event_runner
    handler.run(args[0])
  File "/home/kirika/venv/sucks/lib/python3.7/site-packages/sleekxmpp/xmlstream/handler/callback.py", line 76, in run
    self._pointer(payload)
[... cut ...]
############################## vacbot.run(GetChargeState())
DEBUG:sucks:Sending command <iq id="*" to="*@130.ecorobot.net/atom" from="*" type="set"><query xmlns="com:ctl"><ctl td="GetChargeState" id="*" /></query></iq>
DEBUG:sucks:*** charge_status = charging
############################## vacbot.run(GetBatteryState())
DEBUG:sucks:Sending command <iq id="*" to="*@130.ecorobot.net/atom" from="*" type="set"><query xmlns="com:ctl"><ctl td="GetBatteryInfo" id="*" /></query></iq>
DEBUG:sucks:*** battery_status = 100%
```

bmartin5692, 2019-04-20:
I pushed a new commit in my add-fixStatus branch (https://github.com/bmartin5692/sucks/commit/e6073b5520a5aa082274cbcc2413f3c9c7e68044) that checks for type in the _handle_clean_report, let me know if that helps with the messages.  I was able to add a few more tests from an xmppeek log I have from an Ozmo 930, but I'm not able to reproduce the sleekxmpp errors from the tests.

If you still get those errors we would need xmppeek logs for when this occurs to get more details. See [this](https://github.com/wpietri/sucks/blob/master/developing.md#mitm-xmpp-traffic-between-the-android-or-ios-app-and-the-ecovacs-server) for more details.

Nama, 2019-04-20:
This is, what I get now:
```
DEBUG:sucks:No type in clean report'{'id': '281448135', 'ret': 'ok', 'event': 'clean_report', 'speed': 'standard', 'st': 'h', 't': '100', 'a': '100'}'
```
I'll try to MITM someday, it will take a while till I'm motivated to set that up. I close the issue till then.

Thanks to your example, I can get the status via
```
def subStatusEvent(incomingStatus):
    print('Status: %s' % (incomingStatus))
vacbot.statusEvents.subscribe(subStatusEvent)
```
This even works. But for idle, it returns None.
The status I got from this were: auto, stop, charging and returning

bmartin5692, 2019-04-20:
I just pushed a new commit (https://github.com/bmartin5692/sucks/commit/049cc8d9e0649d8d554b5287be042e22e1201cf2) that I think will address this.  I re-worked how it handles the clean reports checking for a 'st' after a type since the advanced bots seem to send this without types sometimes.  In these cases if the status is 'stopped' or 'paused' those will override the previous 'type' setting.

When you have time try the latest from the add-fixStatus branch (https://github.com/bmartin5692/sucks/tree/add-fixStatus) and let me know if that addresses the remaining issues.

Nama, 2019-04-20:
This looks better. Thank you!
```
DEBUG:sucks:*** clean_status = stop fan_speed = normal
```
How can I use the infos, since the functions are returning None?

bmartin5692, 2019-04-20:
That looks good. I won’t be near a computer for some time, but the functions aren’t meant to return anything just update the statuses for VacBot. 

I pulled this out of the code, you should use these for current status:
        # These three are representations of the vacuum state as reported by the API
        self.clean_status = None
        self.charge_status = None
        self.battery_status = None

        # This is an aggregate state managed by the sucks library, combining the clean and charge events to a single state
        self.vacuum_status = None

Nama, 2019-04-20:
Oh right, I nearly forgot those!
You are the best, thank you for your effort :)

closed 2019-04-20 by Nama