wpietri/sucks #71: Deebot Slim 2 Support?

christophstein, 2019-06-10:
Is it possible to get support for Deebot Slim 2? Currently I don't get through the login process...

bmartin5692, 2019-06-12:
Hi @christophstein -
I believe some other users reported success with the Slim 2 previously.  Could you provide some additional information and logs?  
1. What country are you in?
2. Could you execute sucks with --debug and provide the output?

christophstein, 2019-06-14:
your two-letter country code [de]: de
your two-letter continent code [eu]: eu
sucks      ERROR    call to loginByItToken failed with {'result': 'fail', 'errno': 0, 'error': 'set token error.', 'todo': 'result'}
Traceback (most recent call last):
  File "/usr/local/bin/sucks", line 11, in <module>
    sys.exit(cli())
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 1163, in invoke
    rv.append(sub_ctx.command.invoke(sub_ctx))
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 956, in invoke
    return ctx.invoke(self.callback, **ctx.params)
  File "/home/pi/.local/lib/python3.5/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/home/pi/.local/lib/python3.5/site-packages/sucks/cli.py", line 152, in login
    EcoVacsAPI(device_id, email, password_hash, country_code, continent_code)
  File "/home/pi/.local/lib/python3.5/site-packages/sucks/__init__.py", line 126, in __init__
    login_response = self.__call_login_by_it_token()
  File "/home/pi/.local/lib/python3.5/site-packages/sucks/__init__.py", line 185, in __call_login_by_it_token
    'token': self.auth_code}
  File "/home/pi/.local/lib/python3.5/site-packages/sucks/__init__.py", line 177, in __call_user_api
    "failure {} ({}) for call {} and parameters {}".format(json['error'], json['errno'], function, params))
RuntimeError: failure set token error. (0) for call loginByItToken and parameters {'country': 'DE', 'token': 'de_6eabf21389b081d5941782e95687a28e', 'userId': '20190417214509_caefbf354b4e19c5c2524d2b80ca7d2c', 'resource': 'abbebba7', 'realm': 'ecouser.net', 'todo': 'loginByItToken'}

DFC1990, 2019-06-15:
I get the same error... I can't login ...

bmartin5692, 2019-06-15:
I made some changes in  PR #63  to address this, also available in my master https://github.com/bmartin5692/sucks

bmartin5692, 2019-06-17:
@DFC1990 

Moving the conversation here instead of on PR #63, so it doesn't get too noisy there.  In response to you, I did not see anything out of the ordinary when you were using the sucks.sh method.

I did some testing over the weekend, I'm not sure why the set token error occurs exactly.  I sniffed traffic from the apps and cannot see a difference in how the requests are sent.  It could be an existing token or EcoVacs system being a bit delayed in tracking the new token on their side.

For me if I use portal-eu, I can reproduce this - while portal-ww works every time.  This is why I thought the previous fix was to just use portal-ww, but that doesn't seem to work for you.

I found a method that works for me when testing against portal-eu, by retrying two times if this occurs:
1. Retry as-is 
    - In my testing with portal-eu, the second try would always work and the login would be successful
2. If it fails a second time, then it will retry with portal-ww instead of portal-{continent}
    - I added this as a last-resort
3. If it fails a third time.... Then we'll have to do a deeper analysis of the cause.

I've pushed this change into my [master](https://github.com/bmartin5692/sucks) from a fix_setToken branch.  Please try the changes in my master, and if it works then I will merge into the D900 branch so it ends up in the PR #63.

niveksan, 2019-06-27:
@bmartin5692 

I did try the updated master with OZMO930 to also login from germany via country: de/continent: eu. After starting the session I get a timeout (which wasn't there before these changes but with connecting to us/na instead of de):

`sleekxmppfs.xmlstream.handler.waiter WARNING  Timed out waiting for IqWait_0a42e4f8-3330-40ac-8930-4274667c9ba8-3
Traceback (most recent call last):
  File "/usr/local/bin/sucks", line 11, in <module>
    load_entry_point('sucks==0.9.3', 'console_scripts', 'sucks')()
  File "/usr/local/lib/python3.5/site-packages/Click-7.0-py3.5.egg/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.5/site-packages/Click-7.0-py3.5.egg/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.5/site-packages/Click-7.0-py3.5.egg/click/core.py", line 1164, in invoke
    return _process_result(rv)
  File "/usr/local/lib/python3.5/site-packages/Click-7.0-py3.5.egg/click/core.py", line 1102, in _process_result
    **ctx.params)
  File "/usr/local/lib/python3.5/site-packages/Click-7.0-py3.5.egg/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.5/site-packages/sucks-0.9.3-py3.5.egg/sucks/cli.py", line 231, in run
  File "/usr/local/lib/python3.5/site-packages/sucks-0.9.3-py3.5.egg/sucks/__init__.py", line 623, in run
  File "/usr/local/lib/python3.5/site-packages/sucks-0.9.3-py3.5.egg/sucks/__init__.py", line 617, in send_command
  File "/usr/local/lib/python3.5/site-packages/sucks-0.9.3-py3.5.egg/sucks/__init__.py", line 938, in send_command
  File "/usr/local/lib/python3.5/site-packages/sleekxmppfs-1.3.4-py3.5.egg/sleekxmppfs/stanza/iq.py", line 233, in send
    raise IqTimeout(self)`

btw: I'm also still not able to get back a status from the ozmo (e.g. returning to charger) itself 

drop me a note if you need more from the logs.