wpietri/sucks #73: How to login from Australia

dangerzau, 2019-07-02:
Hi All,

Currently trying to get sucks to work with my 930

I can login via the US settings on such login however when it hen try sucks clean 10 its just errors on and on eabout device unavailable...  

Can someone please provide a step by step on how you got your Ozmo 930 working? as I can't make heads or tails of whats heppening..

<error type="wait" code="404"><recipient-unavailable xmlns="urn:ietf:params:xml:ns:xmpp-stanzas" /></error></iq>

bmartin5692, 2019-07-02:
Hi @dangerzau -

The sucks master doesn't currently support the Ozmo 930.  There are pending pull requests to address this (#63 / #61) .  In the interim you should be able to use the master on my fork:
https://github.com/bmartin5692/sucks/tree/master

dangerzau, 2019-07-02:
Hi Brian, 
I'm pretty sure I am using your master of your fork. 

Is there a special easy I'm supposed to login or a command to run to inform sucks to use a 930?

Also. If I tell sucks login to use the country code au and the other thing it asks for to oc (whatever the defaults are that appear) it will fail login with a server rejected the connection error.

If I use us and na it will at least let me login.

dangerzau, 2019-07-03:
Changed to AU and WW using your repo and that worked a treat... no if only i could get the map data from the deebot sent to openhab as an image.  that would rock

Thankyou Thankyou Thankyou

dangerzau, 2019-07-03:
How does one tell the Deebot to clean the entire house as if I'd just hit the auto button on the app?  I've just initiated an area clean with the % wildcard (I assume its working as a wildcard as the robots vacuming like a madman as we speak)?

I've tried reading through docs however I can't seem to make heads or tails of it all.

Whilst on the subject - And perhaps I should open a new issue as feature request - Would be nice if the vacs status could be pumped out to an mqtt broker. say use a cmnd topic to recieve commands and a status topic to send data such as battery level fan speed etc

bmartin5692, 2019-07-03:
@dangerzau 

I just added this in my master.  If you use `clean 0` it will auto clean until the bot returns to the charger.

https://github.com/bmartin5692/sucks/commit/33d864b3a5cca363a5495174f4fb9960bc7a3b3c

RE: Status-
The sucks CLI is meant to be basic, sucks itself is mainly a library that you can use in your own code.  I know there has been work done to integrate sucks with OpenHAB for example (https://github.com/guillebot/openhab-sucks) that uses MQTT.