wpietri/sucks #76: OZMO 630 Series

codeclinic, 2019-07-14:
I have an OZMO 630 Series vac that I cannot start via sucks. Everything else seems to work fine and sucks is able to see when the device is running already (eg a times or manual switch on/off) but it can't switch the device on. I can send it back to the charger, the charge state shows, I can get it to do the location beeps, and even change fan speed. Howevr, I cannot get it to start the vac running.

bmartin5692, 2019-07-14:
The master here is waiting on PR #63 and PR #61 to work properly with the Ozmo.  In the meantime you could try my fork (https://github.com/bmartin5692/sucks).

codeclinic, 2019-07-14:
I actually meant an OZMO 610 but, hopefully that will work too. Thanks.

grillp, 2019-09-25:
@bmartin5692 , I just tried your Fork on an 610 and it works great! Hope you can get it into master soon. I am using with through http://homeassistant.io!

Great work!

codeclinic, 2019-09-26:
@bmartin5692 I'm not getting any data on the brush & filter life spans when using your fork with an Ozmo 610.

@grillp Are you able to get brush filter data in Hassio ?

grillp, 2019-09-26:
@codeclinic , I never got anything before other than error and sometimes saw it as active, but could not control it. With the this branch I got it working, which is all new to me! So I don't know where I should see the brush and filter data.

codeclinic, 2019-09-26:
@grillp Follow the instructions on Home assistant. Go to the integrations page and look for ecovacs. Everything is/should be the same except you replace "ecovacs" with "ozmo" in your config file.