wpietri/sucks #80: Ozmo950 - Json Communication

bmartin5692, 2019-09-29:
Working on [Bumper](https://github.com/bmartin5692/bumper) I've found that the Ozmo950 and possibly other newer bots are using a new communication method.  All commands to bots are being sent in json instead of xml. 

This is incompatible with Sucks and these bots won't work until a solution is found.

Sucks currently assumes all commands are xml.  It takes commands and wraps them in a json payload for the Ecovacs API, but forces a payloadType "x" for xml as opposed to a "j" for json.

I haven't been able to find a switch or status in order to determine which bot models this would be needed for, but believe it is hard coded in the Ecovacs App based on the "UILogicID" of bots.

Example clean command from Ozmo950:
```json
{
  "body": {
    "data": {
      "act": "start",
      "count": 1,
      "donotClean": 0,
      "router": "plan",
      "type": "auto"
    }
  },
  "header": {
    "pri": 1,
    "ts": "1569380082176",
    "tzm": -240,
    "ver": "0.0.50"
  }
}
```

KoelnSolar, 2019-09-30:
answer to #63 
Yes, I'm able to run the bot with the ecovacs home app 1.2.0(currently I'm installing new version of the 5 days old version). 
With bumper I get access, but no reaction on sending commands. Just the information the bot sends periodically.
Problem here: since I'm a hater off monopolists, I'm running the app in an android emulation environment on  Win-Tablet. Wireshark is installed on another Win-PC and tcpdump on Rpi. I decrypted messages between bumper, app and bot. But how may I do that without having the certificate to decrypt messages between ecovavcs-servers, app and bot ?

gbenrus25, 2019-12-21:
Wondering if there's any update on this? Thanks

KoelnSolar, 2019-12-21:
Still discussed here https://github.com/bmartin5692/bumper/issues/51 and maybe solved for OZMO 950 ?

gbenrus25, 2019-12-21:
Thanks! Asking on that thread for confirmation

Vorms, 2020-05-26:
I am very interested about an integration for the deebot 950 from Ecovacs.
Please, take me informed !

Best regards and many thanks for your work !