wpietri/sucks #84: Could you please make a release?

onkelbeh, 2019-11-22:
Hi,
I would like to use your integration in my [Gentoo Overlay for Home Assistant](https://github.com/onkelbeh/HomeAssistantRepository).
Therefor I need a release file in tar.gz format.
Home Assistant currently requires: `sucks==0.9.4`
This one exists at Pypi, but only in wheel format, cannot use this.
The latest relase on github (in correct format) is 0.9.0

Thanks
 \B.

wpietri, 2019-11-22:
Totally reasonable point. @OverloadUT, It looks like PyPi is at 0.9.4, but what I see in the main branch is 0.9.3. Could you push whatever's in that release to the repo, and then optionally make a release on GitHub as well? Thanks!

onkelbeh, 2019-11-27:
knock - knock. anybody home?

OverloadUT, 2019-11-29:
@wpietri I felt like I was going crazy because I couldn't find any local evidence of posting the 0.9.4 version from my machine. After searching my email, it looks like you were the one that actually published 0.9.4! :)

Could you check your local files for anything unpushed? I'm happy to do the release process on the GitHub side.

If you don't have the local files we can certainly just make the version bump separately as that should be all that is missing

wpietri, 2019-12-01:
Hah! My apologies for dragging you into this. I'll sort this out tomorrow. Thanks for setting me straight!

wpietri, 2019-12-01:
Ok, fixed. It took me a fair bit of archaeology to figure out what was going on, but I believe this is now sorted.

@OverloadUT, part of my mistake here was looking at PyPi and seeing you as possibly having release powers there. Do we need to do anything else, either here or there, for you be able to do releases?

closed 2019-12-01 by wpietri

OverloadUT, 2019-12-01:
@wpietri Nope, I've got all the permissions I need. I actually did publish the 0.9.3 release which is why I also assumed I did the 0.9.4. :)

onkelbeh, 2019-12-02:
Thanks guys,
just added it: https://github.com/onkelbeh/HomeAssistantRepository/commit/727a09e2dd56372c55e0d55e9371832ad04b4024
I promise i won't ask for the older releases ;-)