wpietri/sucks #86: Incorrect Ecovacs continent code for some countries

realbigest, 2019-12-07:
Now Sucks uses pycountry_convert to determine continent code for Ecovacs servers from entered country code. In most cases it’s ok, but there is some exclusions. For example Russia (“ru” country code) now mapping to Europe (“eu” continent code), but Ecovacs apps connecting to “ww” servers.

Maybe we need some dictionary In code for assign known exclusions?

wpietri, 2019-12-09:
I think that's a fine idea. Feel free to put together an exception for just the wrong countries that you know; we can always add more later. But if you're feeling ambitious, I suspect the full list could be extracted from a decompiled version of their Android app.

jyz0501, 2020-03-04:
No CN country code

jyz0501, 2020-03-04:
Keep an eye on this issue

dcwilli, 2021-01-06:
anyone know the Australian continent code?