wpietri/sucks #87: Cannot connect: 404 wait recipient-unavailable xmlns error?

poldim, 2019-12-30:
## Issue summary

- New Deeboot 900
- Connected to WiFi and have no issues controlling the vacuum, both local (WiFi) and remote (4G), through the iOS app.
- I've tried using _na_ and _ww_ continent codes with the same result.
- Might be related to #85 

## Home Assistant
Openning the vacuum entity and issuing a command gives the same error:
`Failed to call service vacuum/stop. <iq type="error" to="dkisml171b2b9d21@ecouser.net/0402ca35962e558a173df28518e8d88a" from="bf4c4e7f-a453-4d55-ad80-fd4e172c5271@ls1ok3.ecorobot.net/atom" id="b7c9afd4-5b8c-4105-84eb-a70042493743-2"><error type="wait" code="404"><recipient-unavailable xmlns="urn:ietf:params:xml:ns:xmpp-stanzas" /></error></iq>`

Home Assistant log reports:
```
Ping did not reach VacBot. Will retry.
9:41 AM components/ecovacs/vacuum.py (WARNING) - message first occurred at 9:36 AM and shows up 12 times
```

## Using CLI to connect:
```
bash-5.0# sucks login
Ecovacs app email: ****@gmail.com
Ecovacs app password:
your two-letter country code [us]: us
your two-letter continent code [na]: na
Config saved to /root/.config/sucks.conf
bash-5.0# sucks clean 10
sleekxmppfs.basexmpp WARNING  fulljid property deprecated. Use boundjid.resource
performing clean command
Traceback (most recent call last):
  File "/usr/local/bin/sucks", line 8, in <module>
    sys.exit(cli())
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 764, in __call__
    return self.main(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 717, in main
    rv = self.invoke(ctx)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1164, in invoke
    return _process_result(rv)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 1102, in _process_result
    **ctx.params)
  File "/usr/local/lib/python3.7/site-packages/click/core.py", line 555, in invoke
    return callback(*args, **kwargs)
  File "/usr/local/lib/python3.7/site-packages/sucks/cli.py", line 219, in run
    vacbot.run(action.vac_command)
  File "/usr/local/lib/python3.7/site-packages/sucks/__init__.py", line 416, in run
    self.send_command(action.to_xml())
  File "/usr/local/lib/python3.7/site-packages/sucks/__init__.py", line 413, in send_command
    self.xmpp.send_command(xml, self._vacuum_address())
  File "/usr/local/lib/python3.7/site-packages/sucks/__init__.py", line 482, in send_command
    c.send()
  File "/usr/local/lib/python3.7/site-packages/sleekxmppfs/stanza/iq.py", line 235, in send
    raise IqError(result)
sleekxmppfs.exceptions.IqError: <iq type="error" to="dkisml171b2b9d21@ecouser.net/0402ca35962e558a173df28518e8d88a" from="bf4c4e7f-a453-4d55-ad80-fd4e172c5271@ls1ok3.ecorobot.net/atom" id="2ce2fddc-103e-4617-8732-a70042493743-2"><error type="wait" code="404"><recipient-unavailable xmlns="urn:ietf:params:xml:ns:xmpp-stanzas" /></error></iq>

^C
Exception ignored in: <module 'threading' from '/usr/local/lib/python3.7/threading.py'>
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/threading.py", line 1307, in _shutdown
    lock.acquire()
KeyboardInterrupt
bash-5.0#
```

bmartin5692, 2019-12-30:
For a 900, you'll need to use my fork at this time - https://github.com/bmartin5692/sucks

poldim, 2019-12-30:
> For a 900, you'll need to use my fork at this time - https://github.com/bmartin5692/sucks

Thanks for the help!  Do you have any insight of if it's possible to force the home assistant container to install your fork?  Or any hopes of merging it into the main HA repo?

Also, I tried to run your bumper project with docker compose but running into an issue where the same docker host is currently also running my mosquitto server so I can't start it due to a port conflict.  I tried to set a static IP for bumper from my host network, but the 900 didn't recognize it.  If you have any recommendations here, they are welcome!

bmartin5692, 2019-12-30:
For HA, I'm not sure. I believe some of the folks on the mailing list have forked my sucks for use in HA, you may want to check history there.

Bumper runs its own MQTT Server, so you would need to use a macvlan network to provide a new IP address is my guess.

closed 2020-05-25 by wpietri