wpietri/sucks #91: Deebot 600 on HA not working

sunsgs, 2020-04-13:
Hi all, i have set in my configuration.yaml

```
ecovacs:
  username: !secret ecovacs_username
  password: !secret ecovacs_password
  country: it
  continent: eu
```

but it keeps saying 
`failed vacuum/turn_on . <iq type="error" to="dlekk1h99a53a0ed@ecouser.net/e74810235bd8c5ebb7645276bc0acf00" from="f4bd7659-4e1f-4522-a1bd-92f68f4f4793@dl8fht.ecorobot.net/atom" id="38f54846-b1b8-4914-a6fd-4dd6a73e8d3d-6"><error type="wait" code="404"><recipient-unavailable xmlns="urn:ietf:params:xml:ns:xmpp-stanzas" /></error></iq>`

any help appreciated

mbaehr2909, 2020-05-03:
Did you try setting the Continent to "WW"?

duncan-bayne, 2020-05-24:
I've tried setting Continent to WW (using the Home Assistant integration), but this problem persists. 

I've contacted Ecovacs Australia to request API docs to help debug the problem.

duncan-bayne, 2020-06-16:
The API docs request is progressing ... will attempt to debug this issue once I've received them.

Shadowbird21, 2020-08-16:
Hey, is there anything new about this? I'm having the same issue with my Deebot 600 in Germany.
Here is my configuration.yaml and the error log, maybe they help?
```
ecovacs:
  username: !secret ecovacs_user
  password: !secret ecovacs_password
  country: de
  continent: eu
```
`Fehler beim Aufrufen des Service vacuum/turn_on. <iq type="error" to="ebzp093107cbda31@ecouser.net/2e8316f95d178026afdb23236ae577bb" from="319c83fd-3a03-499d-bf29-e99cd77cd5ca@dl8fht.ecorobot.net/atom" id="6e601a54-9c39-41f6-ba86-a5087be80cda-471"><error type="wait" code="404"><recipient-unavailable xmlns="urn:ietf:params:xml:ns:xmpp-stanzas" /></error></iq>` (The German part just says error executing service)

wpietri, 2020-08-16:
I don't think the 600 was ever known to work, and nobody is currently working on this project, so I'd suggest you look elsewhere.

Shadowbird21, 2020-08-17:
Ok, thanks.

genericnerd99, 2020-08-20:
im seeing the same with my 920 tried the diffrent codes as well, im in the UK

Shadowbird21, 2020-08-21:
After spending quite a bit of time sniffing all the traffic that the app sends when logging in and executing the commands, and comparing this to the library, I believe the issue may be that Ecovacs changed the base URL for commands on some devices (at least for my 600 model)