wpietri/sucks #93: Request for official Ecovacs API docs

duncan-bayne, 2020-05-24:
After a [surprisingly helpful response](https://community.home-assistant.io/t/rinnai-heating-cooling-wifi-module/109908/213?u=duncan_bayne) from Rinnai to my request for their API docs, I've emailed EcoVacs to request their API docs, so I can help debug #91 on my DeeBot600.

I'll post updates to this issue as I hear back from Ecovacs.

duncan-bayne, 2020-05-24:
Received the following automated acknowledgement from EcoVacs support:

```
Your request (13706) has been received and is being reviewed by our support staff.
```

duncan-bayne, 2020-05-24:
Snipping irrelevant bits, the conversation continues:

```
Greetings from Ecovacs.

Thank you for contacting Ecovacs. Kindly please allow us to explain that the Ecovacs Deebot can only support Amazon Alexa and Google home. We would like to thank you for the feedback given to us, rest assured that we will escalate this matter to our relevant department. 

We are more than happy to assist you with anything you may need and look forward to with our products and services. Kindly let us know if you have any more questions, comments or concerns. Thank you for support. 
```

```
Thanks for getting back to me so quickly. 

I understand that Ecovacs only supports Google and Amazon. 

I'm asking that you provide documentation for your API, so that I can complete the Home Assistant integration without having to reverse engineer your protocol. 

If you're unable to provide the documentation, could you please escalate to one of your engineering teams who can? Thank you. 
```

No need to be disheartened at this point; it took a month of this sort of conversation to get traction with Rinnai :)

DendelX, 2020-05-24:
Ecovacs has one of the worst customer support services I have ever seen. Just to answer an email it takes months. I have had problems for 2 times with a deebot 930. One of them was inventing non-existent carpets and the other was not able to return to its charging base.

Both returned thanks to Amazon, because they did not answer any of my emails. They came to see him when he had already returned both devices.

Not to mention that in press releases they promised a multi-plant for the 930 and then they did not fulfill it. I can only recommend running away from this brand ...

wpietri, 2020-05-25:
For what it's worth, I talked with the Ecovacs head of US support last year and was not able to get docs, so I wouldn't be surprised if they say no. And from the quality of the API, I'd expect that they may not actually have any docs. But there's no harm in trying, and I appreciate your optimism.

duncan-bayne, 2020-05-25:
The conversation continues :). Apparently there's an open ticket already, which ... seems to have been merged into my request?

```
Request #13706 "API support" was closed and merged into this request. 
```

I've asked for clarification:

```
Thanks - glad to see there's already an API support ticket open.
Could you please let me know what the status of that ticket its?  
```

🤞 it's making some progress.

duncan-bayne, 2020-05-26:
Looks like it might be:

```
Thank you for contacting Ecovacs. We would like to thank you for the feedback given to us, rest assured that we will escalate this matter to our relevant department. We are more than happy to assist you with anything you may need and look forward to with our products and services. Rest assured once we have the outcome we will update you accordingly.
```

My reply, just to ensure this isn't a boilerplate "thank your for your feedback" ...

```
Hi :). Thanks - I'm not quite sure what you mean by escalate, though.  Will someone be in touch with the API documentation?
```

It looks like they're actually following up on it:

```
Thank you for contacting Ecovacs. Kindly please note that once we receive further information in regards to API Documentation query from the relevant team, we will update you accordingly.
```

... but I suspect I'll need to chase it up through other channels to keep it moving.

duncan-bayne, 2020-06-06:
Looks like nothing will be forthcoming; I've followed it up on the support ticket:

```
It's been a while and I haven't had any reply.

Could you please confirm that this is being worked on, and whether
there's an ETA?

Other people are reporting having contacted Ecovacs some time ago, but
no API documentation has ever been released.
```

... and will be chasing alternative channels (e.g. contacts on LinkedIn).

duncan-bayne, 2020-06-06:
Have reached out to the head of AU / NZ on LinkedIn (I'm in Australia).  Will see if I get anywhere.

duncan-bayne, 2020-06-09:
Getting nowhere with the support team.  Hoping to get further through management channels.

```
> Thank you for contacting Ecovacs. We have forwarded your request about 
> the API to our Relevant Team and we regret to inform you that we are 
> unable to provide API documentation on further checking with the 
> relevant team. We apologize for the inconvenience caused.

May I ask why?
```

duncan-bayne, 2020-06-10:
Ah!  Perhaps words have been had.

```
Thank you for contacting Ecovacs. We apologize for the previous email and we need to escalate further with the Ecovacs team. Kindly provide the required details as per below to proceed further to escalate your request with the Ecovacs team.
 
Kindly furnish us with the following details to proceed.
```

I'll provide them with all the details they require (pertaining to the models in question) and see where we get to.

duncan-bayne, 2020-06-16:
Details provided - serial number, model number, proof of purchase, the whole nine yards.

duncan-bayne, 2020-06-19:
> Please be advised that we are currently looking into this matter. We would like to suggest you try again within 5-7 business days.  Rest assured once we have the outcome we will update you accordingly.

Will follow up ...

duncan-bayne, 2020-06-25:
Radio silence ... have given them a gentle prod.

```
It's been 7 days but I haven't heard anything back.  Are you making
progress on obtaining the documentation?
```

duncan-bayne, 2020-06-26:
Not promising.

```
As per further checking, our relevant team is requesting the Photo of error code screenshot to be sent to further assist. Kindly please provide the requested photo for us to forward to the relevant team to further assist you. 
```

Will provide them the screenshot from Home Assistant showing the error, though.

duncan-bayne, 2020-06-26:
```

I'm not sure how helpful the screenshot will be.

I'm asking for the API documentation, so that I can extend the 'sucks' library to support the DeeBot 600.

Nonetheless, I've included a screenshot of the error message from Home Assistant in the hopes it's helpful.
```

duncan-bayne, 2020-07-12:
```
Thank you for the email. We regret to inform we unable to provide the code due to the confidentiality. Please be advised that the Ecovacs Deebot can only support Amazon Alexa and Google home. We would like to thank you for the feedback given to us
```

I've explained the difference between source code and an API, and await another, probably useless, response.

duncan-bayne, 2020-10-13:
Turns out Ecovacs is unwilling to provide the documentation:

> We understand your concern and have relay the information however your request for documentation denied due to confidentiality.

closed 2020-10-13 by duncan-bayne

OidaTiftla, 2020-10-15:
Seems they follow the not so recommended concept of `security by obscurity` 😄