wpietri/sucks #96: How to logon from the UK?

collse, 2020-07-25:
I have tried various things and could identify the following:

  country: gb
  continent: eu

seems to be the right option to chose but the endpoint in the library is set to: eco-gb-api.ecovacs.com (which doesnt not resolve)

when trying to redirect it via dns to the endpoint (eco-uk-api.ecovacs.com) that actually replies to ping (unlike the eco-gb-api.ecovacs.com ) this then at least starts the authentication process but fails with:

2020-07-25 09:45:40 DEBUG (SyncWorker_6) [sucks] got {'code': '0001', 'msg': '操作失败', 'time': 1595666740099, 'data': None} - google translate this says operation failed
2020-07-25 09:45:40 ERROR (SyncWorker_6) [sucks] call to user/login failed with {'code': '0001', 'msg': '操作失败', 'time': 1595666740099, 'data': None}

as the endpoint eco-gb-api.ecovacs.com doesnt seem to exist and the ww endpoint doesnt work either I wonder what may have changed here?

authentication via app with same credentials works just fine for my N79S.

genericnerd99, 2020-08-16:
Hey - just to say im seeing the same in the UK, i have tried multiple variations of uk,gb,eu,ww with same results, pings to uk resolve as above as well.  also seeing:
Logger: sucks
Source: /usr/local/lib/python3.8/site-packages/sucks/__init__.py:371
First occurred: August 16, 2020, 11:57:34 PM (7 occurrences)
Last logged: 12:00:34 AM

Ping did not reach VacBot. Will retry.
Thanks for any help

Shadowbird21, 2020-08-21:
Check out #91. I believe I found the problem (at least for my Deebot 600) in Germany.

leelfd, 2020-10-14:
I am in the UK and had exactly the same issue so I corrected the convert_country_alpha2_to_continent_code.py script to replace the GB code with UK. Now it works.

Shadowbird21, 2020-10-16:
> I am in the UK and had exactly the same issue so I corrected the convert_country_alpha2_to_continent_code.py script to replace the GB code with UK. Now it works.

Do you know where I can change that, if I want to use the Ecovacs Home Assistant integration? Can I find that file using the File Editor?